opendata-stuttgart/sensors-software #2: How to get more than 100mA from an USB port

asmaps, 2015-06-09:
Find out what part we need to add to tell the other end that we will draw more than 100mA from the USB port.

http://electronics.stackexchange.com/questions/5498/how-to-get-more-than-100ma-from-a-usb-port

asmaps, 2015-06-09:
http://www.aliexpress.com/item/Genuine-FT232RL-FT232-SSOP-28-100-Brand-New-and-Original/1611426733.html

relatively expensive, but would directly include serial communication.

armin-bauer, 2015-06-09:
There's a lot of Mainboards that only have USB Ports that don't care about the 100mA limitation and even supply you with more than 500 mA . 
If you need rs232 anyway however, you could check out the CH340G which is afaik the cheapest chip available with 20 to 40 ct per piece.

Marwe, 2016-11-26:
closing: nodemcu dev boards are used, with everything on board...

closed 2016-11-26 by Marwe