opendata-stuttgart/sensors-software #4: opensensors.io as mqtt broker

tiborb, 2015-10-18:
Opensensors is a convenient and free way to make the particle data open/public.
The account holder can publish to a topic, topics can be public or private, public topic can be used by anyone. 

Usage:
```
mosquitto_pub -h mqtt.opensensors.io -i <client_id> -t <topic> -u <dusty> -m 'measue and timestamp' -P <your password> 
```

Use cases:
1. Sensors publish directly to opensensors (permissions must be reviewed)
2. Only dusty publishes to opensensors

tiborb, 2015-12-10:
@mfa The MQTT for ESP8266 is available at https://github.com/tiborb/ESP8266/tree/master/mqtt_arduino/mqtt_esp8266

In which sensors-software (arduino) variant should I start integrating it?

ensonic, 2017-01-14:
There are some basics already here:
https://github.com/opendata-stuttgart/sensors-software/tree/master/esp8266-arduino/ppd42ns-wificonfig-ppd-sds-dht

ricki-z, 2017-01-14:
this is an very old issue. a new version will be published in the next days, needs some more testing by now. this version will have mqtt support.

ensonic, 2017-01-14:
I just implemented mqtt support as well :/

https://github.com/ensonic/sensors-software/commit/6024144359112330897e758f46d19d44f0fcdb47

Is this like what you have (which branch?)? Or do you want a PR for this?

It works, but I think the graph feature on opensensors.io is broken (or I haven#t figured how to get graphs from my data). As a general improvement it would be nice to store number in json as numbers and have the units as a separate field (optional).

ricki-z, 2017-01-15:
please see #33. We will remove mqtt completely from our firmware.

closed 2017-01-15 by ricki-z