opendata-stuttgart/sensors-software #29: Sensor restarts and looses Wifi

frankgerhardt, 2017-01-09:
My sensor (301) reboots frequently. I checked the power and think it's not a power issue. It works fine after a reboot but the Wifi config is lost. 

Is there a way to configure the SSID and Wifi password ***persistently*** so that I don't have to go through the manual configuration each time?

Is there a script to configure the sensor so that I don't have to go through the manual configuration each time?

fuergestalten, 2017-01-11:
To solve the problem with the fluctuation of current just try another power plug.

ricki-z, 2017-01-13:
Hi Frank,
some NodeMCUs seem to have problems with the flash rom. The SPIFFS isn't working, so the config can't be saved. You should see this, if you look at the serial debug output. You could try to write an empty SPIFFS (as written in our docu under "delete the config").
To configure the sensor persistently, you have to compile the firmware with the correct options set in ext_def.h. And you have to disable auto updates.
Any time the sensor restarts, it will look for the config.json. If this isn't found, the compiled predefined values will be used. So if you use auto update this would be "Freifunk" for SSID with no password set.

ensonic, 2017-01-14:
I can find any docs that tell how to "delete the config" either.
```
find . -type f -exec grep -Hn "delete the config" {} \;
```
I've tried to implement the missing pieces for MQTT (see #4), but now I need to override "send2mqtt":false in the stored config.

ricki-z, 2017-01-14:
You need to delete the config.json in SPIFFS, this is described in https://github.com/opendata-stuttgart/meta/wiki/Firmware-einspielen at the end of the page.
In the next firmware version there will be mqtt, the posibility to delete the config and a better configuration via browser.
If you have special requirements on mqtt, please comment here (message format, mqtt server ...)

ricki-z, 2017-01-15:
@ensonic : please don't hijack unrelated threads.

closed 2017-06-04 by ricki-z