opendata-stuttgart/sensors-software #33: Work in progress: support for InfluxDB and MQTT as backend, new WiFiConfig

ricki-z, 2017-01-14:
These are some enhancements that will come with the next versions:

support for InfluxDB as backend
support for MQTT as backend

new config page as a more flexible replacement for WiFiManager

ensonic, 2017-01-15:
What do you mean by "too many different payload strings and auth methods"? If this gets dropped, whats the suggested way to log the data to some place where I can get history and e.g. graphs?

ricki-z, 2017-01-15:
Some mqtt brokers need user name and password, others need a device id and a device password.
I've requests for a payload with a special format and a payload with json formatted data.
You can easily implement a "proxy", that translate json to mqtt. Then you can use this proxy as custom api.
Historic data can be found here: http(s)://archive.luftdaten.info/
"Life" data can be found here: http(s)://api.luftdaten.info/static/v1/data.json . This file is updated every minute and contains all sensors and values sent to the server in the last 5 minutes.

ensonic, 2017-01-15:
Thanks for the explanation, makes sense. I'll look into the custom-api suggestion. Is there an irc / gitter.im chat for this project (so that I don't hijack more threads :))?

ricki-z, 2017-01-15:
We use slack. Please send a short mail to our info address with the request for an invitation.

amotl, 2017-05-04:
Dear Raijko and the whole team of luftdaten.info,

first things first: Thanks for conceiving this great project!

@einsiedlerkrebs and me recently started to write a small program for doing the live json api to MQTT convergence like outlined in your post above, see [luftdaten-to-mqtt]. It is currently used to feed data from the [luftdaten.info live api] to an instance of our [Kotori data historian] in order to display historical data from all sensors by geographical location using a rich data visualization dashboard in [Grafana] to make it a meaningful supplement to the [luftdaten.info map].

Maybe this can spark your interest. To get more into the details, you might want to have a look at [about luftdaten.info in Grafana]. We are happy to receive any feedback and answer further questions. Feel free to ask us anything :-).

Cheers,
Andreas.

[luftdaten-to-mqtt]: https://github.com/daq-tools/kotori/blob/master/kotori/vendor/luftdaten/luftdaten_api_to_mqtt.py
[about luftdaten.info in Grafana]: https://getkotori.org/docs/applications/luftdaten.info/
[Kotori data historian]: https://github.com/daq-tools/kotori
[Grafana]: https://github.com/grafana/grafana
[luftdaten.info map]: http://deutschland.maps.luftdaten.info/
[luftdaten.info live api]: https://api.luftdaten.info/static/v1/data.json

njumaen, 2017-05-05:
@amotl This ist soooooo awesome!!! Great work!

tho-wa, 2017-05-21:
Is it down actually @amotl ?

amotl, 2017-05-22:
Dear Thomas (@tho-wa),

you are right, very sorry. We expanded the disk size of the machine before the weekend and missed to spin up the (separate) Grafana instance again. Now its up and running again, thanks for noticing!

Cheers,
Andreas.

ricki-z, 2019-02-01:
@rzr please read the complete thread. There are good reasons why we don't support MQTT (memory usage, different configs for different platforms, and if we start with one platform how should we argue against others)

rzr, 2019-06-17:
Well i assumed there were traction for MQTT support:

* https://github.com/opendata-stuttgart/sensors-software/issues/4
* https://github.com/opendata-stuttgart/sensors-software/issues/5
* https://github.com/opendata-stuttgart/sensors-software/issues/113
* https://github.com/opendata-stuttgart/sensors-software/pull/165
* https://github.com/opendata-stuttgart/sensors-software/issues?utf8=%E2%9C%93&q=mqtt

So I guess it should be implemented in per platform custom fw, 

I am targeting this device:
ESP8266 ESP8266EX LoLin NodeMcu v3

Any clue ?

ricki-z, 2019-06-17:
You could modify the "Own API" functionality. Or set up a mapper JSON to MQTT and use this as an "Own API" endpoint.
There is more than just implementing MQTT. This protocol is very unspecific about the data format.
And: The source code should work on all ESP8266 systems with more than 1MB flash memory. There is no need for a per platform fw.

dirkmueller, 2019-11-28:
Looks like nothing is still open here  mqtt had been removed (although it could be added back now that we have fewer constraints on memory)

closed 2019-11-28 by dirkmueller