opendata-stuttgart/sensors-software #35: DHT22 is not read

njumaen, 2017-01-25:
h = dht.readHumidity(); //Read Humidity  
    t = dht.readTemperature(); //Read Temperature

and connecting to 5V (Vin) instead of 3,3V

solved the issue for me!

Using DHT Sensor Library by Adafruit Version 1.3.0

ricki-z, 2017-01-25:
Hi, njumaen,
this is correct. We will change our description, so that the DHT22 is also connected to 5V.
But you can use the Vin only on NodeMCU v2. On newer boards there is an diode, that prevents the usage of Vin as Vout. So you need to build a cable that will power both the SDS011 and the DHT22.
By now I haven't found the change in the DHT Library since our firmware version NRZ-2016-039. Until this version the DHT22 was working as expected.
In our firmware we force the read with dht.readHumidity(true) and dht.readTemperature(false,true) (false for degree Celcius). The firmware seems to use the same variable name as the the library to save the last reading. So after compilation the firmware won't show the correct readings and our graphs show many gaps.
So please try the actual firmware version NRZ-2016-046 which should solve most of the problems.

njumaen, 2017-01-25:
Using NodeMCU v2 I just did so... 

NRZ-2016-046 binary has this issue. So I just got the .ino-source and the latest libraries and built my own... Same issue. 
Then I tried the DHTtester.ino from the library examples. There are just () in the functions calls für °C ! Worked! My DHT22 is OK!
Next I changed your code and *woooosh* it worked.
#33c3 "works for me"   Check out sensor 15858932 !

njumaen, 2017-01-25:
// Example testing sketch for various DHT humidity/temperature sensors
// Written by ladyada, public domain

#include "DHT.h"

#define DHTPIN 2     // what digital pin we're connected to

// Uncomment whatever type you're using!
//#define DHTTYPE DHT11   // DHT 11
#define DHTTYPE DHT22   // DHT 22  (AM2302), AM2321
//#define DHTTYPE DHT21   // DHT 21 (AM2301)

// Connect pin 1 (on the left) of the sensor to +5V
// NOTE: If using a board with 3.3V logic like an Arduino Due connect pin 1
// to 3.3V instead of 5V!
// Connect pin 2 of the sensor to whatever your DHTPIN is
// Connect pin 4 (on the right) of the sensor to GROUND
// Connect a 10K resistor from pin 2 (data) to pin 1 (power) of the sensor

// Initialize DHT sensor.
// Note that older versions of this library took an optional third parameter to
// tweak the timings for faster processors.  This parameter is no longer needed
// as the current DHT reading algorithm adjusts itself to work on faster procs.
DHT dht(DHTPIN, DHTTYPE);

void setup() {
  Serial.begin(9600);
  Serial.println("DHTxx test!");

  dht.begin();
}

void loop() {
  // Wait a few seconds between measurements.
  delay(2000);

  // Reading temperature or humidity takes about 250 milliseconds!
  // Sensor readings may also be up to 2 seconds 'old' (its a very slow sensor)
  float h = dht.readHumidity();
  // Read temperature as Celsius (the default)
  float t = dht.readTemperature();
  // Read temperature as Fahrenheit (isFahrenheit = true)
  float f = dht.readTemperature(true);

  // Check if any reads failed and exit early (to try again).
  if (isnan(h) || isnan(t) || isnan(f)) {
    Serial.println("Failed to read from DHT sensor!");
    return;
  }

  // Compute heat index in Fahrenheit (the default)
  float hif = dht.computeHeatIndex(f, h);
  // Compute heat index in Celsius (isFahreheit = false)
  float hic = dht.computeHeatIndex(t, h, false);

  Serial.print("Humidity: ");
  Serial.print(h);
  Serial.print(" %\t");
  Serial.print("Temperature: ");
  Serial.print(t);
  Serial.print(" *C ");
  Serial.print(f);
  Serial.print(" *F\t");
  Serial.print("Heat index: ");
  Serial.print(hic);
  Serial.print(" *C ");
  Serial.print(hif);
  Serial.println(" *F");
}

njumaen, 2017-01-25:
Result:

DHTxx test!
Humidity: 89.00 %	Temperature: 22.90 *C 73.22 *F	Heat index: 23.57 *C 74.43 *F
Humidity: 69.80 %	Temperature: 23.60 *C 74.48 *F	Heat index: 23.84 *C 74.91 *F
Humidity: 66.10 %	Temperature: 23.70 *C 74.66 *F	Heat index: 23.85 *C 74.93 *F
Humidity: 63.40 %	Temperature: 23.80 *C 74.84 *F	Heat index: 23.89 *C 75.00 *F
Humidity: 61.20 %	Temperature: 23.90 *C 75.02 *F	Heat index: 23.94 *C 75.10 *F
Humidity: 57.00 %	Temperature: 24.10 *C 75.38 *F	Heat index: 24.05 *C 75.30 *F
......

ricki-z, 2017-01-25:
Works for you, but not for anyone ...
I've tested the version before 046 and everything worked fine for my two sensors.
The "true" in the function calls forces a value read. Otherwise the function will use saved values, if the last reading was within the last 2 seconds.
The false in the dht.readTemperature() is needed to be able to use the second parameter. Its the default, so without the forced reading ist the same as an empty function call.
With that change we could get 50 percent of the sensors with problems back to work. The others may need to get connected to the 5V Vout.
Look at the following sensors:
esp8266-13597771, DHT connected to 5V, latest firmware:
https://www.madavi.de/sensor/graph.php?sensor=esp8266-13597771-dht
esp8266-742561, DHT connected to 5V, latest firmware:
https://www.madavi.de/sensor/graph.php?sensor=esp8266-742561-dht

For this sensor, DHT22 connected to 3.3V, the latest firmware solved the problem with false readings: https://www.madavi.de/sensor/graph.php?sensor=esp8266-2338440-dht

But for others like yours the latest firmware won't read any DHT22 values.

So my solution isn't working for all, your solution won't work for all. Maybe something between those two should solve the problem.

I will rewrite the function this way, that there is first an "unforced" reading. And only if this doesn't work a second, forced reading is done.

The changed firmware will be available today in the late evening.

njumaen, 2017-01-25:
hardware dependencies. bad thing.

ricki-z, 2017-01-26:
Is solved in version NRZ-2017-047.

closed 2017-01-26 by ricki-z