opendata-stuttgart/sensors-software #38: Grafik: Verschaltung Sensor

polygontwist, 2017-02-02:
Hallo, im Wiring-pdf ist VUSB mit 2,5µ statt 5V vom Feinstaubsensor (SDS011) verbunden - ist das korrekt? In der Textlichen Beschreibung ist er hingegen auf 5V gelegt und der 2,5µ nicht benutzt (halte ich für richtiger)!
(https://github.com/opendata-stuttgart/sensors-software/tree/master/esp8266-arduino/ppd42ns-wificonfig-ppd-sds-dht)

ricki-z, 2017-02-02:
Das ist richtig. Es sollte auf den 5V-Pin gehen. Martin, kannst du das korrigieren?

Marwe, 2017-02-03:
Asked ET to provide update from source

rabede, 2017-02-26:
RXD und TXD sind ebenfalls vertauscht!

// SoftSerial RX PIN is D1 and goes to SDS TX
// SoftSerial TX PIN is D2 and goes to SDS RX

Marwe, 2017-02-26:
Changed PDF (with LO)

closed 2017-02-26 by Marwe