opendata-stuttgart/sensors-software #41: ArduinoJson 5.8.2 Version geht nicht, 5.7.3 und 5.8.1 ist OK

polygontwist, 2017-02-02:
aktuelle Version von ArduinoJson (5.8.2) funktioniert nicht mit Projekt, die 5.7.3 geht, die 5.8.1 compiliert auch.

**5.8.2 Fehlermeldungen:**
> Build-Optionen wurden verändert, alles wird neu kompiliert
> C:\Users\andreas\Documents\GitHub\sensors-software\esp8266-arduino\ppd42ns-wificonfig-ppd-sds-dht\ppd42ns-wificonfig-ppd-sds-dht.ino: In function 'String create_influxdb_string(const String&)':
> 
> C:\Users\andreas\Documents\GitHub\sensors-software\esp8266-arduino\ppd42ns-wificonfig-ppd-sds-dht\ppd42ns-wificonfig-ppd-sds-dht.ino:1401:88: warning: 'const char* ArduinoJson::JsonVariantBase<TImpl>::asString() const [with TImpl = ArduinoJson::JsonObjectSubscript<const char*>]' is deprecated (declared at C:\Users\andreas\Documents\Arduino\libraries\ArduinoJson/include/ArduinoJson/Deserialization/../JsonVariantBase.hpp:36): use as<char*>() instead [-Wdeprecated-declarations]
> 
>     tmp_str = jsonBuffer.strdup(json2data["sensordatavalues"][i]["value_type"].asString());
> 
>                                                                                         ^
> 
> C:\Users\andreas\Documents\GitHub\sensors-software\esp8266-arduino\ppd42ns-wificonfig-ppd-sds-dht\ppd42ns-wificonfig-ppd-sds-dht.ino:1403:83: warning: 'const char* ArduinoJson::JsonVariantBase<TImpl>::asString() const [with TImpl = ArduinoJson::JsonObjectSubscript<const char*>]' is deprecated (declared at C:\Users\andreas\Documents\Arduino\libraries\ArduinoJson/include/ArduinoJson/Deserialization/../JsonVariantBase.hpp:36): use as<char*>() instead [-Wdeprecated-declarations]
> 
>     tmp_str = jsonBuffer.strdup(json2data["sensordatavalues"][i]["value"].asString());
> 
>                                                                                    ^
> 
> C:\Users\andreas\Documents\GitHub\sensors-software\esp8266-arduino\ppd42ns-wificonfig-ppd-sds-dht\ppd42ns-wificonfig-ppd-sds-dht.ino: In function 'void send_csv(const String&)':
> 
> C:\Users\andreas\Documents\GitHub\sensors-software\esp8266-arduino\ppd42ns-wificonfig-ppd-sds-dht\ppd42ns-wificonfig-ppd-sds-dht.ino:1431:88: warning: 'const char* ArduinoJson::JsonVariantBase<TImpl>::asString() const [with TImpl = ArduinoJson::JsonObjectSubscript<const char*>]' is deprecated (declared at C:\Users\andreas\Documents\Arduino\libraries\ArduinoJson/include/ArduinoJson/Deserialization/../JsonVariantBase.hpp:36): use as<char*>() instead [-Wdeprecated-declarations]
> 
>     tmp_str = jsonBuffer.strdup(json2data["sensordatavalues"][i]["value_type"].asString());
> 
>                                                                                         ^
> 
> C:\Users\andreas\Documents\GitHub\sensors-software\esp8266-arduino\ppd42ns-wificonfig-ppd-sds-dht\ppd42ns-wificonfig-ppd-sds-dht.ino:1433:83: warning: 'const char* ArduinoJson::JsonVariantBase<TImpl>::asString() const [with TImpl = ArduinoJson::JsonObjectSubscript<const char*>]' is deprecated (declared at C:\Users\andreas\Documents\Arduino\libraries\ArduinoJson/include/ArduinoJson/Deserialization/../JsonVariantBase.hpp:36): use as<char*>() instead [-Wdeprecated-declarations]
> 
>     tmp_str = jsonBuffer.strdup(json2data["sensordatavalues"][i]["value"].asString());
>

ricki-z, 2017-02-02:
Ich habe hier die Arduino IDE 1.8.1, ArduinoJSON 5.8.2 und auch alle anderen Bibliotheken auf aktuellem Stand. Kompilieren läuft durch ohne Meldung.

polygontwist, 2017-02-03:
bis auf die ArduinoJson ist alles bei mir auch aktuell:
ArduinoJson 5.8.1
Adafruit Unified Sensor 1.0.2 (aktuellste)
Adafruit BMP085 library 1.0.0 (aktuellste)
Adafruit BME085 library 1.0.4 (aktuellste)
DHT sensor library 1.3.0 (aktuellste)
ESP8266 Oled driver for SSD1306 display 3.2.5 (aktuellste)
ESP8266httpUpdate 1.1.0 (aktuellste)
PubSubClient 2.6.0 (aktuellste)
SoftwareSerial  1.0.0 (aktuellste)
WifiManager 0.12.0 (aktuellste)
TinyGPS++ 0.94b (aktuellste)

die Arduino IDE ist die aktuelle 1.8.1 neu installiert.
nunja zumindest geht's bei mir mit der 5.8.1.

pj4y, 2017-02-06:
@polygontwist 
ich denke Du meinst Adafruit BME280 library 1.0.4 anstatt Adafruit BME085 library 1.0.4?

Ansonsten habe ich nun die gleiche Konfiguration. Die fragliche ArduinoJson 5.8.2 hatte ich ohnehin schon ohne Probleme im Einsatz.

Und ja, ich erhalte die **warning**s auch. Dazu muss ich aber die Compiler-Warnungen hochschrauben. Es sind halt Warnungen, dass die eingesetzten Deklarationen deprecated sind. Funktionieren tut es (zumindest bei mir) natürlich dennoch. 

Man sollte das also entweder für zukünftige Versionen Zwecks Kompatibilität fixen oder halt die letztmalig zu nutzende Version dokumentieren und das irgendwann mal fixen, wenn man den Schritt zu einer höheren Version macht, die es dann über Board wirft.

**geht nicht** ist vielleicht etwas hart formuliert.

polygontwist, 2017-02-08:
> ich denke Du meinst Adafruit BME280 library 1.0.4 anstatt Adafruit BME085 library 1.0.4?
ja, sorry das habe ich beim abschreiben übersehen.

Ich sehe es aus der Sicht des unbedarften Anwender und bin darüber gestolpert. Wenn die Warnungen nichts bedeuten kannst Du mein Kommentar gerne ignorieren. 
Ich kenne andere Projekte die einen "stabilen" Zweig und einen "Entwicklungszweig" führen - da kommen solche Irritationen nicht auf, bzw. wenn man die Entwicklungsversion nimmt kann man damit rechnen. Evtl. ist das eine Idee? So kann man in Ruhe neue Funktionen austesten und weiterentwickeln.

Danke für Deine Arbeit.

closed 2017-02-08 by polygontwist

ricki-z, 2017-02-08:
Ich habe gerade eine neue Version gepusht, bei der die warnings nicht mehr kommen.

Als kleiner Kommentar:
Die unbedarften Nutzer nehmen den kurzen Weg über die fertige Firmware-Datei.
Wer den Quellcode nutzt, sollte schon mal mit der Arduino (oder einer anderen) IDE gearbeitet haben und wissen, was eine Warning ist. Ich kenne nur ganz wenige Projekte, bei denen das Kompilieren ganz ohne Warnings auskommt. Nimm dir mal aus Spaß einen Linux-Kernel aus dem Stable-Zweig und laß den mal übersetzen. Dort hast du Warnings, mit denen du einige Tage beschäftigt sein dürftest, wenn du die alle lesen willst.
Und spätestens, wenn das Hochladen auf die NodeMCU funktioniert, ist auch mit dem Code soweit alles in Ordnung.
Der Code im Repository ist getestet. Vorher wird dieser nicht veröffentlicht.
Nochmal: Warnings sind keine Fehler.

polygontwist, 2017-02-09:
Also für mich was der Weg über Arduino einfacher als eine bin zu nutzen. Die Projekte die ich bisher mit Arduino gehackt habe, hatten keine Meldungen (auch kleine ESP8266 Projekte). Und es war ja _keine_ Warnung unterm Strich, der Compiler hat ja **abgebrochen** und ich musste eine andere Version probieren. Ich dabei hatte ich ja alles neu installiert (inkl. Arduino).
Aber nun funkte es ja fröhlich vor sich hin.
:-)