opendata-stuttgart/sensors-software #48: Using kadamski/arduino_sds011 library

corny, 2017-02-10:
This library could help to clean up the code:
https://github.com/kadamski/arduino_sds011/tree/master/lib/Sds011

Your opinion on this?

ricki-z, 2017-02-10:
We need some parts of this code (start cmd, stop cmd, ...) in other scripts and sources. Here we have a place to look for them. Some users want to use the SDS011 on Raspberry Pis or other systems. I can't find these cmds in the mentioned source (okay, may be it's to late to look for something like this ...).

corny, 2017-02-10:
The commands are included in the `Sds011.h`. I use the library this way:
```c
SoftwareSerial serialSDS(SDS_PIN_RX, SDS_PIN_TX, false, 128);
sds011::Sds011 sensorSDS(serialSDS);

void initSDS(){
  sensorSDS.set_sleep(false);
  sensorSDS.set_mode(sds011::QUERY);
  sensorSDS.set_sleep(true);
}
```

ricki-z, 2017-02-10:
We need the serial data that ist sent to the SDS011 to start and stop. Not the library commands.
And as far as I can see, these commands are not "active" (CMD_SLEEP ist not set).

corny, 2017-02-10:
Alle the commands have values. It's a `ENUM` and they get their values implicitly by incrementing the previous value.

ricki-z, 2017-02-10:
Okay, but our intention stays the same. Having the possibility to copy the binary command code to other "languages".

ricki-z, 2017-02-10:
This will only clean up our code. The size of the firmware file or the used mem will stay nearly the same.

corny, 2017-02-10:
I don't see a valid reason for that. In other languages there are much easier ways to implement it than in C++.

* Python: http://www.codegists.com/snippet/python/sds011_kadamski_python
* Ruby: http://www.codegists.com/snippet/ruby/sds011rb_qoobaa_ruby
* other random language

ricki-z, 2017-02-10:
see changes in last comment. WiFiManager is another lib we had to rewrite for our project even if for slightly other reasons.

corny, 2017-02-12:
In my opinion the current SDS011 code bloat your code and is hard to read and understand. What do you think about copying the library into your repository?

ricki-z, 2017-02-12:
Sorry, the SDS011 code will be left there.
(160 out of 2300 lines of code)

closed 2017-02-12 by ricki-z