opendata-stuttgart/sensors-software #49: Anzeige der MAC-Adresse

polygontwist, 2017-02-11:
Hallo, gäbe es die Möglichkeit auf noch die MAC-Adresse anzuzeigen. Ich hab in meinem WLAN einen MAC-Filter und es wäre schön wenn man diesen bei der (erst) Konfiguration sehen könnte.

Anosnten biete ich hier noch ein Beispielsketch zum auslesen der MAC-Adresse:
```
/*
 *  This sketch demonstrates how to scan WiFi networks. 
 *  The API is almost the same as with the WiFi Shield library, 
 *  the most obvious difference being the different file you need to include:
 */
#include "ESP8266WiFi.h"

uint8_t MAC_array[6];
char MAC_char[18];

void setup() {
  Serial.begin(115200);

  // Set WiFi to station mode and disconnect from an AP if it was previously connected
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);

  WiFi.macAddress(MAC_array);
  for (int i = 0; i < sizeof(MAC_array); ++i){
      sprintf(MAC_char,"%s%02x:",MAC_char,MAC_array[i]);
    }

  Serial.println(MAC_char);
  Serial.println("Setup done");
}

void loop() { 
}
```

ricki-z, 2017-02-11:
Kommt mit dem nächsten Update und ist dann im Header zu sehen. Aktuelle Version ist gerade im Test. Wenn alles funktioniert, sollte die Version morgen live gehen und auf die Sensoren verteilt werden.

corny, 2017-02-12:
Funktioniert bei mir. Issue kann geschlossen werden?

closed 2017-02-12 by ricki-z