opendata-stuttgart/sensors-software #52: Angabe von GPS-Koordinaten

corny, 2017-02-12:
Es sollte möglich sein, bei der Konfiguration die GPS-Koordinaten einzugeben, damit diese zusammen mit den Messwerten übertragen werden. Ähnliches machen z.B. die Freifunker in ihrer Gluon-Software.

Das vereinfacht bei euch ebenso die Verwaltung der Positionen von Sensoren.

njumaen, 2017-02-12:
Fänge ich gut. Vor Allem wenn man den Sensor umzieht... Oder: Optional die GPS-Position in die Datenbank mitübertragen. Dann könnte man mittels mobile Sensoren Daten sammeln

ricki-z, 2017-02-12:
Wir haben aktuell andere Prioritäten als mobile Sensoren. Und für die geplanten Auswertungen reichen uns die reinen GPS Daten nicht. Wir benötigen genauere Angaben über das "Umfeld" des Sensors. Wir müssen also mehr als die GPS-Koordinaten anpassen. Die Verwaltung der Sensoren wird damit nicht wirklich einfacher.

corny, 2017-02-12:
Wie wird das Umfeld konkret in der Datenbank abgebildet?
> Email an [...] mit den Daten: z. B. Höhe über Grund, Strassenseite, hohes Verkehrsaufkommen, freies Feld oder ähnliches

Wenn ihr die Daten wirklich manuell per E-Mails verwalten wollt, wird das auf Dauer ganz schön viel Arbeit...

ricki-z, 2017-02-12:
Es wird an einem Formular gearbeitet, das dann die Angaben in die API schreibt. Der Sensor muss dann nur noch freigegeben werden, damit der Sensor auf der Karte erscheint.
Es gibt leider genug Beispiele, bei denen eine "zu einfache" Eintragung zu völlig abstrusen Einträgen führt. Deshalb werden wir die manuelle Freigabe beibehalten.

corny, 2017-02-12:
Danke für die Erklärung.

closed 2017-02-12 by corny