opendata-stuttgart/sensors-software #53: HTTP Header umbenennen

corny, 2017-02-12:
`Pin` und `Sensor` sind keine standardisierten HTTP-Header. Sie sollte daher umbenannt werden in `X-Pin` und `X-Sensor`. Wofür wird der Pin-Wert überhaupt verwendet?

ricki-z, 2017-02-12:
Dafür müssten wir den API-Server anpassen. Dieser erwartet den Pin-Header, um die Werte einem Sensortyp zuordnen zu können. Zudem existieren inzwischen zahlreiche API-Clients, welche diese Header nutzen und nicht so ohne weiteres geupdated werden können. Wir müssten also auf lange Zeit die API jeweils 2 Varianten auswerten lassen.

corny, 2017-02-12:
Ok, verstanden.

ricki-z, 2017-02-12:
Ich habe mir gerade

ricki-z, 2017-02-12:
Ich habe mir den Code für die Header-Auswertung gerade nochmal angesehen. Ich werde den so umbauen, das beide Varianten (also mit und ohne 'X-') möglich sind. Damit können dann die 'alten' Clients weiterhin ihre Daten schicken.

corny, 2017-02-12:
Sehr schön!

ricki-z, 2017-02-12:
Ist geändert, schau mal auf die aktuelle Version.

corny, 2017-02-13:
Sieht gut aus :+1:

closed 2017-02-13 by corny