opendata-stuttgart/sensors-software #54: Messerwerte bündeln?

corny, 2017-02-12:
Kann ich die Daten mehrere Sensoren (z.B. SDS011 und BME280) in einem Request an api.luftdaten.info übertragen? TLS-Handshakes sind ziemlich teuer - das macht sich auch auf eurem Server bemerkbar.

Beispiel:
```json

{
  "software_version": "NRZ-2016-054",
  "sensordatavalues": [
    {
      "value_type": "P1",
      "value": "11.25"
    },
    {
      "value_type": "P2",
      "value": "10.12"
    },
    {
      "value_type": "temperature",
      "value": "22.70"
    },
    {
      "value_type": "humidity",
      "value": "34.57"
    },
    {
      "value_type": "pressure",
      "value": "103294.48"
    }
  ]
}
```

ricki-z, 2017-02-12:
Ganz kurz: nein. Es findet eine Prüfung der Daten auf dem Server statt. Deren Komplexität und damit zeitlicher Aufwand wäre deutlich größer als für den TLS Handshake.

corny, 2017-02-12:
Ok, wo finde ich den Server-Code?

closed 2017-02-12 by corny

ricki-z, 2017-02-12:
https://github.com/opendata-stuttgart/feinstaub-api

corny, 2017-02-12:
danke

ricki-z, 2017-02-12:
Kleine Ergänzung: Es gibt Sensoren mit überschneidenden Werten. So hat der DHT22 Temperatur u. Luftfeuchte, der BMP180 Temperatur und Luftdruck, der BME280 alle 3 Werte. Da wir jeweils die verwendeten Sensoren pro Node kennen, können wir prüfen, ob die Werte passen.
Aktuell geben also Pin-Header und Sensor-ID gibt die zu erwartenden Werte vor. Die Datenbankabfrage dafür "kostet" relativ wenig.
Beim Senden aller Daten auf einmal müssen wir alle Sensoren je Node zusammensuchen und die möglichen Werte prüfen.
Alternativ könnte man das JSON noch eine Ebene mehr schachteln und eine Sensor-Ebene einfügen. Größere JSON Strukturen werden aber auf dem ESP8266 schnell kritisch im Bezug auf den Speicher. Kommt dann noch die Verschlüsselung drauf, hat man schnell eine abstürzende NodeMCU. Das mussten wir schon leidvoll erfahren. Wir sind deshalb etwas knausrig mit neuen Funktionen, damit Heap und Stack möglichst viel Platz haben.
In der aktuellen Version ist es uns z.B. deshalb nicht möglich, Zertifikate größer 2048 Bit für den API-Server zu verwenden.

corny, 2017-02-12:
Problematisch sehe ich eher, dass das generierte JSON unter Umständen erneut auf dem ESP8266 geparst wird. Die Messwerte sollten erst mal zwischengespeichert werden, bevor sie serialisiert werden. Beispiel aus meinem Code: https://gist.github.com/corny/23ee508606bb7dd3579bfaba849230ba#file-outdoor-ino-L47

Kleine Idee für ein alternatives und kompakteres JSON-Format:
```json
{
  "software_version": "NRZ-2016-054",
  "sensordatavalues": {
    "SDS011": {
      "P1": 11.25,
      "P2": 10.12
    },
    "DHE280": {
      "temperature": 22.70,
      "humidity": 34.57,
      "pressure": 103294.48
    }
  }
}
```
oder:

```json
{
  "software_version": "NRZ-2016-054",
  "sensordatavalues": [
  {
    "type": "SDS011",
      "P1": 11.25,
      "P2": 10.12
    },
    {
    "type": "DHE280",
      "temperature": 22.70,
      "humidity": 34.57,
      "pressure": 103294.48
    }
  ]
}
```

ricki-z, 2017-02-12:
Das Parsen auf der NodeMCU habe ich schon versucht zu minimieren. Es gibt für einige Werte schon globale Variablen, die die letzten Werte zwischenspeichern. Aber hier haben wir das Problem des gewachsenen Codes. Ich schaue bei Gelegenheit nochmal drüber.
Das JSON für die API sollte eigentlich nicht mehr lokal geparst werden müssen. Leider ist das Format durch Django auf dem API-Server vorgegeben.

ricki-z, 2017-02-14:
Ich habe jetzt nochmal nachgeschaut, wo wir das Daten-JSON parsen. Ich sehe da eher Vorteile in der jetzigen Version. send_csv und create_influxdb_string müssen so beide nicht "wissen", welche Sensoren vorhanden sind. Bei neuen Sensoren muss ich die beiden Funktionen auch nicht nochmal anfassen.

corny, 2017-02-14:
Ich glaube, du hast mich nicht verstanden, oder meinen Code nicht gelesen? Nach dem Refactoring sieht das so aus bei mir:
```c

struct {
	bool valid;
	float t; // Temperature
	float h; // Humidity
	float p; // Pressure
} bme280_result = {};

void readBME280() {
	bme280_result.t = bme280.readTemperature();
	bme280_result.h = bme280.readHumidity();
	bme280_result.p = bme280.readPressure();
	bme280_result.valid = !isnan(bme280_result.t) && !isnan(bme280_result.h) && !isnan(bme280_result.p);
}

void readAndSend(){
	if (bme280_read)
		readBME280();
	else
		bme280_result.valid = false;

	// blah ...

	if (bme280_result.valid) {
		data += Value2Json(F("temperature"), bme280_result.t);
		data += Value2Json(F("humidity"),    bme280_result.h);
		data += Value2Json(F("pressure"),    bme280_result.p);
	}

	// und abschicken
}
```

corny, 2017-02-14:
Also: Erfassung der Messdaten und Serialisierung voneinander trennen. Macht den Code lesbarer und vermeidet Probleme wie generiertes JSON zu parsen.