opendata-stuttgart/sensors-software #56: ESP freeze after refreshing values page

Cash70, 2017-02-25:
The ESP will freeze (Firmware Version: NRZ-2016-062) if the values http page was updated (refresh button at the browser) while the firmware is communicating with the SDS011 sensor (led on the sensor flashing).
If this occurs, the ESP is was completely dead, no more http communication, no ping possible (timeout).
If the web pages are not ussed, the ESP works like a charm.

Last log lines at log level 5 are at every freeze:
...
End connecting to api.luftdaten.info
## Sending to luftdaten.info (DHT): 
Start connecting to api.luftdaten.info
Requesting URL: /v1/push-sensor-data/
1197671
{"software_version": "NRZ-2016-062", "sensordatavalues":[{"value_type":"temperature","value":"22.10"},{"value_type":"humidity","value":"35.60"}]}

njumaen, 2017-02-28:
got issue with latest ppd42ns-wificonfig-ppd-sds-dht.ino.nodemcu.bin

Starting WiFiManager
AP ID: Feinstaubsensor-663780
output not found page...
output not found page...
output not found page...
output not found page...
validate request auth...
output root page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
validate request auth...
output root page...
output luftdaten.info logo...
output codefor.de logo...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
validate request auth...
output config page ...
scan for wifi networks...
wifi networks found: 5
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...

Exception (28):
epc1=0x4000bdc8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff2d10 end: 3fff30d0 offset: 01a0

>>>stack>>>
3fff2eb0:  00000000 3fff13a0 3fff2ed0 4021aca4  
3fff2ec0:  00000000 3fff13a0 3fff137c 402149e6  
3fff2ed0:  00000000 00000000 00000000 3fffbcb4  
3fff2ee0:  0000005f 00000058 3fffbbe4 0000000f  
3fff2ef0:  0000000a 3fff6344 0000000f 00000000  
3fff2f00:  3fff632c 0000000f 00000000 3fffbbac  
3fff2f10:  0000002f 00000025 3fff72b4 0000002f  
3fff2f20:  00000004 3fff692c 0000000f 00000004  
3fff2f30:  00000000 000002cf 000002cf 40211464  
3fff2f40:  00000000 3fff1c4c 00000130 00000000  
3fff2f50:  00000000 40212ac8 0000000a 402113ee  
3fff2f60:  00000000 3fff0fe8 3fff137c 3fff1fcc  
3fff2f70:  000005b4 3fff13a0 3fff137c 40213a8b  
3fff2f80:  3ffe9a38 00000000 000003e8 4021298d  
3fff2f90:  00000000 3fff8d0c 3fff1db8 4021baa8  
3fff2fa0:  3ffe934e 3fff0fe8 3fff1db8 4020a9c4  
3fff2fb0:  3ffe9af8 00000000 000003e8 000004bc  
3fff2fc0:  00000000 3fff4504 3ffe0035 3fff3ab4  
3fff2fd0:  0000000f 00000001 0104a8c0 00000000  
3fff2fe0:  00000000 3fff8d0c 3c000000 00000000  
3fff2ff0:  00000000 00000000 00000000 3ffe0035  
3fff3000:  0104a8c0 3ffea2d0 0104a8c0 3ffea2d0  
3fff3010:  00ffffff 3ffea2d0 0104a8c0 4010068c  
3fff3020:  3ffe9710 3ffe934e 3fff3040 3fff20a8  
3fff3030:  3ffe934e 3ffe934e 3fff1db8 4020d61c  
3fff3040:  00000000 00000000 00000000 4010068c  
3fff3050:  3fff0ff8 3fff1fcc 3fff3070 4021aca4  
3fff3060:  3fff0ff8 3fff1fcc 00002580 4020e0b5  
3fff3070:  00000000 00000000 00000000 00000000  
3fff3080:  00000000 00000000 feefeffe feefeffe  
3fff3090:  feefeffe feefeffe feefeffe feefeffe  
3fff30a0:  feefeffe feefeffe feefeffe 3fff20a8  
3fff30b0:  3fffdad0 00000000 3fff20a0 4021ba74  
3fff30c0:  feefeffe feefeffe 3fff20b0 40100718  
<<<stack<<<
.......DH...mounting FS...
mounted file system...
reading config file...
opened config file...
parsed json...
Starte Webserver... 0.0.0.0
0
Connecting to njumaen
....................
Starting WiFiManager

njumaen, 2017-02-28:
built my own .bin with latest libs... same issue :-(

saeugetier, 2017-03-11:
Got the same error. I used https://www.madavi.de/sensor/update/data/latest.bin

ricki-z, 2017-03-11:
The issue is known. The NodeMCU will crash if a sensor web page is called while the firmware is sending data to our servers. We are working on it an will mark this issue solved as soon as we have a solution.
So please don't sent any more "me too"'s.

ricki-z, 2017-05-31:
Actual versions should stop the internal webserver before sending data. So the nodemcu won't crash anymore at this point.

closed 2017-05-31 by ricki-z