opendata-stuttgart/sensors-software #57: Ambiguous specification of Arduino IDE to install

dhebbeker, 2017-03-07:
Different Arduino IDE versions are stated in the installation guides. This can be confusing.
* [Beginner's Guide](https://github.com/opendata-stuttgart/sensors-software/blob/master/BeginnersGuide/Guide.md#board-manager-installieren) line 23 (last changed by @noobilicious) version _1.6.5_ is specified. The link given _references explicitly the [old software releases page](https://www.arduino.cc/en/Main/OldSoftwareReleases#previous)_.
* [ppd42ns-wificonfig-ppd-sds-dht/Readme.md](https://github.com/opendata-stuttgart/sensors-software/blob/master/esp8266-arduino/ppd42ns-wificonfig-ppd-sds-dht/Readme.md#ben%C3%B6tigte-software) line 40 (last changed by @ricki-z) states that version _1.8.1_ is required
* [meta/wiki/Firmware-einspielen](https://github.com/opendata-stuttgart/meta/wiki/Firmware-einspielen#firmware-einspielen-und-konfig-l%C3%B6schen) (last changed by @ricki-z) no version is specified at all

**Is it required to use a specific version of Arduino IDE?** Otherwise I would like to suggest to soften the specification: In cases where a version is specified, add a statement like 
> (getestete Version)

In case this change is approved, I volunteer to implement it.

ricki-z, 2017-03-08:
In Readme.md I've inserted a statement, that versions in brackets are tested versions.
No versions in Beginner's Guide (versions removed, links changed) and meta/wiki/ should be okay.

dhebbeker, 2017-03-09:
* [Beginner's Guide](https://github.com/opendata-stuttgart/sensors-software/blob/master/BeginnersGuide/Guide.md#board-manager-installieren) has been fixed in 753791c8a907bee5a4cc7877a848a1b07073ff39
* [ppd42ns-wificonfig-ppd-sds-dht/Readme.md](https://github.com/opendata-stuttgart/sensors-software/blob/master/esp8266-arduino/ppd42ns-wificonfig-ppd-sds-dht/Readme.md#ben%C3%B6tigte-software) has been fixed in b7abff2c1bfdc209cea9734fcb803cafb85a5b91

closed 2017-03-09 by dhebbeker