opendata-stuttgart/sensors-software #59: Sensor config page availability for reconfiguration

maehw, 2017-03-20:
Hi there,
I have cloned the sensors software from this repository.

I have compiled and flashed the firmware in /esp8266-arduino/ppd42ns-wificonfig-ppd-sds-dht/ successfully for my NodeMCU.

The sensor's WiFi for configuration came up once, as described in http://luftdaten.info/feinstaubsensor-bauen/. I have set the credentials to connect to my own WiFi. As I wanted to take another look at the sensor's "website" within the config WiFi (to check for valid sensor data) and I tried to reflash the Arduino sketch.

I cannot find the "Feinstaubsensor-*" SSID again. Is this also covered by "Bitte beachten: Wenn die Konfiguration des Sensors funktioniert hat, ist diese Seite nicht mehr erreichbar", i.e. are the credentials stored in a part of the flash memory not overwritten by the sketch and therefore expected functionality? If not, can you give some hints on how to debug this issue, please?

Cheers,
maehw

ricki-z, 2017-03-20:
As soon as the wifi connection is working, the hotspot is closed. But you can access the sensor webpage again if you use the IP given to the sensor by your router.
In some cases http://feinstaubsensor-<chipid>.local will also work.

maehw, 2017-03-21:
Thank you for your reply. I didn't know the webserver on the sensor would still be active after WiFi configuration, thanks for the hint. Still wondering where the WiFi config itself is stored, if not overwritten by flashing the sketch again. Some other non-volatile memory in the NodeMCU?

ricki-z, 2017-03-21:
The 4MB flash memory ist splitted in 1MB for firmware and 3MB for SPIFFS. This could be changed to 3MB/1MB, but we have some sensors with 1MB/1MB flash. And the config is saved as a json file in the SPIFFS part. This part won't be overwritten by updates.
So we don't need a firmware file per sensor and can write a more general firmware.
Here is a short description of OTA updates: https://github.com/esp8266/Arduino/blob/master/doc/ota_updates/readme.md#updater-class

maehw, 2017-03-21:
Thank you for the clarification! ;)

closed 2017-03-21 by maehw