opendata-stuttgart/sensors-software #64: Localisation question

darcoto, 2017-04-11:
Hi,
Do you have any plans for multi-language web interface of ESP?

I know that I can fork repo, but it is not very useful to  keep it in sync with master only for translation.
I hope you will choose some technology for multi-language strings in sources.
If you are agree that it must be done and give some ideas, I can take some work to implement it.

Best regards

ricki-z, 2017-04-11:
We can't save too many language versions in the firmware. The FlashROM and RAM ist very limited. We don't have any idea by now how to solve this.

darcoto, 2017-04-19:
I made some tests to find more effective way to solve. I try some different approach (with ROM and RAM storage) and my conclusion is that it is not very effective to store more then one language in FlashROM without unnecessarily space lost. 
In real usage people never use more than one language in there devices.
So I can offer solution that will build N-firmware files (where N=number of translated languages) each one contains core functions + one language. 
Every language will be one extra .h file and compiler will make N-.bin files with suffix like xxx_de.bin, xxx_en.bin
Auto update must send request with parameter for current language and download relevant file.

I can make changes and suggest pull request, but first of all I would like to know your opinion - whether this approach acceptable for you ?

Best regards

ricki-z, 2017-05-08:
Can you give me the command line to compile the firmware with platform.io?

hakan42, 2017-05-08:
pio run

Or, if you want to upload the generated binary to an esp

pio run -t upload

hakan42, 2017-05-08:
This reminds me that I wanted to write documentation for using platform.io

ricki-z, 2017-05-08:
I'm getting errors on pio run: .platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/nodemcuv2_bg/firmware.elf section `.text' will not fit in region `iram1_0_seg'
There seems to be a problem with the actual version.

darcoto, 2017-05-09:
Hmm,
That's strange.
Can you give me some information about build environment ? 
Currently I can not reproduce it.
I successfully build 3 firmwares on Mac OS and on Ubuntu with latest version of platformio. Ububtu installation is fresh and clear.   

You can compile single firmware with command
`pio run -e nodemcuv2_de`
`pio run -e nodemcuv2_en`
`pio run -e nodemcuv2_bg`

Try to find is the problem persist in all three version or only in bulgarian.
There are almost equal except that in intl_bg.h have a little bit lager svg logo (I will make it smaller). 
You can try to replace it with original for the test.

ricki-z, 2017-05-09:
It was a fresh install on MacOS 10.11.6 as written on platform.io. The problem persists in all three versions.
My steps:
installed platform.io
deinstall python-dateutil, install python-dateutil==2.2
run 'platformio run'

tjuerges, 2017-05-09:
I can verify that the behaviour that @ricki-z has seen is genuine and not a one-off problem.

It is not unique to this project and has already been brought up in the espressif8266 firmware issues here [section '.text' will not fit in region 'iram1_0_seg'](https://github.com/platformio/platform-espressif8266/issues/4).  The issue was closed because a workaround was found although it has not been understood what the real root of the problem is.

ricki-z, 2017-05-09:
This solved the problem (important: don't 'sudo'):
platformio platform uninstall espressif8266
platformio platform install https://github.com/platformio/platform-espressif8266.git#feature/stage
Set platform = espressif8266_stage in platformio.ini

closed 2017-05-31 by ricki-z