opendata-stuttgart/sensors-software #66: Aussetzer bei der Übertragung

ricki-z, 2017-04-15:
Einige Sensoren haben regelmäßige, kurze Aussetzer bei der Übertragung der Werte.
Beispiele:
1010, 1011
1044, 1045
1517, 1518

Wer auch solche Aussetzer z.B. in den Grafiken hat, bitte kurz melden. Am Besten mit Angaben zum Internetanschluß: Provider, Geschwindigkeit des Ansschlusses, Router-Modell.

TheNitek, 2017-04-15:
Ich hatte immer wieder Aussetzer. Grund war, dass der Sensor am Rande des Empfangsbereichs meiner Fritz!Box steht (Smartphone verliert die Verbindung auch ab und an). Habe die Fritz!Box etwas besser positioniert und seit dem sieht es (fast immer) gut aus.

ricki-z, 2017-04-15:
HIer geht es eher um Sensoren, die eine "WLAN-Qualität" von 50% und besser haben. Wir übertragen mit den Daten auch die Signalstärke, können daher in Grenzfällen auch mal einen Hinweis bezüglich WLAN-Empfang geben.
Die angesprochenen Sensoren haben also trotz guter WLAN-Verbindung Probleme, die Daten an unseren Server zu senden.

bgoerner, 2017-04-26:
Das hat sich m.E.n. deutlich verbessert? Auch der Aufruf der aktuellen Werte, etc.

ricki-z, 2017-04-27:
Aktuelle Änderungen:
- der interne Webserver ('aktuelle Werte',...) wird während des Sendens der Daten deaktiviert.
- Verlängerung des Sendeintervalls auf 150 Sekunden, dadurch hoffentlich längere Lebensdauer des SDS011
- Es wird zur Übertragung Port 80 genutzt, wenn das WLAN verschlüsselt ist. HTTPS braucht eigentlich zuviel RAM für einen stabilen Vertrieb.

Das entsprechende Update ist auf dem Weg auf die Sensoren. Damit sollte das dann noch etwas stabiler laufen.

reibuehl, 2017-04-27:
Welche (auf dem Sensor sichtbare) build oder Versionsnummer wäre das?

ricki-z, 2017-04-27:
Die Versionsnr ist NRZ-2017-078 . Upload ist gestern mittag erfolgt, die Sensoren sollten alle 24 Stunden danach schauen. Manchmal bricht der Download ab, dann wird 24 Stunden später noch ein Versuch gestartet.

hakan42, 2017-04-27:
Meine Box hat heute früh das Update bekommen, anschließend war es wieder im AP-Mode, und hatte alle Konfigurationen vergessen (die Defaults waren selektiert). Ist das ein bekanntes Problem?

ricki-z, 2017-04-27:
@hakan42 Wie ist denn die Signalstärke am Sensor?
Siehe dafür http://www.madavi.de/sensor/signal.php (suchen nach der ID der NodeMCU)
Bisher haben meines Wissens nach alle Sensoren nach dem Update die Konfiguration übernommen (aktuell sind ca. 600 aktualisiert und deutlich über 600 aktiv).
Wenn der Sensor den WLAN-Router nicht erreichen kann, geht er in den AP-Mode.

hakan42, 2017-04-27:
Eigentlich primastens, der Sensor steht vielleicht fünf Meter von dem Access Point ( Unifi UAP-AC-Lite ) entfernt. Das einzige was mir einfällt ist, daß es je nach Tageszeit durch die Glasfenster und die Rolläden hindurchfunken muß. Plastik sollte aber nicht allzusehr dämpfen, oder?

Hier mein Sensor:

https://www.madavi.de/sensor/signal.php?sensor=esp8266-1447335

ricki-z, 2017-04-27:
Das scheint zu passen. Kann es sein das der Router weg war, während der Sensor neu gestartet ist?
Wir vermuten bei manchen (zum Glück recht wenigen) NodeMCUs aber auch Probleme mit dem Flash. Dann klappt wohl etwas mit dem Schreiben der Konfig nicht, obwohl ein Okay zurückkommt.

hakan42, 2017-04-28:
Dann riecht es bei mir ehrlich gesagt nach dem Flash :(

Es war ja nicht nur der Sensor im AP-Mode, auch alle settings waren zurück auf den defaults aus ext_def.h.

Mal schauen, ich behalte das mal im Auge, und wenn der nächste update wieder die settings tötet wird es Zeit für einen neuen ESP...

dokape, 2017-09-16:
Wie ist der Zustand bei den betroffenen Nodes?
Ist das inzwischen besser?

@hakan42 hast Du deine Herausforderung lösen können? War es der Flash-speicher? 

Besteht noch weiterer Diskussionsbedarf oder kann es geschlossen werden?

closed 2018-06-09 by ricki-z