opendata-stuttgart/sensors-software #67: Add Sensor ID to sensordatavalues for own API

gregorwolf, 2017-04-20:
I think it would be a great addition to the JSON data that is sent in the sensordatavalues to provide also the Sensor ID. If you have multiple sensors you currently have to use URL Parameters or use different endpoints for the different sensors.

darcoto, 2017-04-27:
You can find Sensor ID in HTTP Header of request. Look for X-Sensor

`POST /data.php  HTTP/1.1`
`Host: api.host.com`
`X-Sensor: esp8266-1335052`

gregorwolf, 2017-04-29:
Good to know that the data is available in the HTTP header. But for the ease of consumption i.e. with Node-Red it would help to have it in the JSON directly.

darcoto, 2017-05-25:
If @ricki-z confirm, I can add necessary code and make PR to project.
But in my opinion, this will add a few line of code witch will duplicate existing information (in header). As you may know, ESP has a limited memory and it is not a good idea to waste it.

ricki-z, 2017-05-25:
I've added the ID to the JSON sent to custom api in the actual version. This will add 112 Byte to filesize. can someone test it?

rexfue, 2017-05-26:
I've tested it. It doesn't work as is.
You send 'data' but should send 'data_4_custom'.
So change line 2573 to:
sendData(**data_4_custom**,0,host_custom,port_custom,url_custom,basic_auth_custom.c_str(),FPSTR(TXT_CONTENT_TYPE_JSON));

Then it works as expected .

ricki-z, 2017-05-26:
Is changed. Will be pushed to the sensors in the next days.

gregorwolf, 2017-05-26:
Excellent. Thank you all. Will check and then close the issue.

gregorwolf, 2017-06-03:
Got Firmware: NRZ-2017-086 on my sensors now and the JSON does now contain esp8266id

closed 2017-06-03 by gregorwolf