opendata-stuttgart/sensors-software #71: Einheitendiskrepanz Luftdruck zwischen luftdaten und sensemap

hakan42, 2017-04-27:
Hallo,

Ich habe in meiner Sensorbox zusätzlich  eine BME280 verbaut und lasse die Daten auch an die opensensemap weiterleiten. 

Box: esp8266-1447335

Gegenstelle: 
http://opensensemap.org/explore/58fa358b63572200114d6b97

Dabei ist mir aufgefallen dass die Box den Druck in Pascal anzeigt und an die sensemap schickt, die sensemap allerdings mit hPa rechnet.

Tschüß,
Hakan

dokape, 2017-09-18:
Sehe ich richtig, dass das immer noch aktuell ist?

hakan42, 2017-09-18:
Ja, das ist immer noch ein aktuelles Problem.

dokape, 2017-09-19:
@ricki-z 
Wenn ich das richtig sehe, wird bei der Übergabe an opensensemap der Luftdruck um das Hunderfache zu groß angegeben. Das sollte doch recht einfach zu ändern sein?

ricki-z, 2017-09-19:
Wir lesen und speichern die Rohdaten der Sensoren. Eine Nachbearbeitung findet erst auf den Servern statt. So können wir im Nachhinein Fehler in Berechnungen korrigieren.
Die Luftdrucksensoren geben richtigerweise Pascal aus. Die Einheit Hektopascal hat sich etabliert, weil die Werte sonst immer recht groß wären, bei KiloPascal aber wohl auch wieder zu klein.
Der Sensor liefert Pascal, deshalb wird auch dieser Wert an die APIs weitergegeben.

dokape, 2017-09-20:
Bei opensensemap kann man Pascal sowie Hektopascal als Einheit einstellen. 
Bei Pascal ist der Wert  korrekt.
Eine Umrechnung in hPa scheint bei opensensemap nicht vorgegeben zu sein. Dort wird der Wert wie angeliefert angezeigt. Pascal ist zwar eher ungewöhnlich, aber nicht falsch. 
Dann ist die Folgerung: Wer hPa wünscht, muss einen Zwischenschritt wie eigener Server à la Raspi einbauen, der aufbereitet und weitergibt. 

Weitere Diskussion nötig oder kann das issue geschlossen werden?
2do: Hinweis entsprechend in Dokumentation aufnehmen.

DeeKey, 2019-12-30:
Is this still an issue after years?

hakan42, 2020-01-01:
Leider, und traurigerweise, ja.

Ich würde ja verstehen dass ich einen Proxy dazwischenschalten soll, wenn ich irgendetwas "eigenes" tun würde.

Allerdings ist die Option, Daten an OpenSenseMap zu senden, bereits in die Firmware eingebaut ("More API" auf der Konfigurationspage), daher würde ich diese simple division um 100 auch in der Firmware etwarten...

Tschüß,
Hakan

tom-r, 2020-01-26:
Hab' das gleiche Problem, nervt.
Werd' wohl 'forken' müssen ...

Gruß,
Thomas

ricki-z, 2020-01-26:
Habt ihr schon mal daran gedacht, bei opensensemap nachzufragen, ob dort nicht zentral alle eingehenden Luftdruck-Werte von Luftdaten.info-Sensoren durch 100 geteilt werden können?

tom-r, 2020-01-27:
Hi,
ja hab ich (mittels der angegebenen email support@...).
Hab aber bisher keine Antwort bekommen.
Und mit Javascript kenne ich mich zu wenig aus, um das selber fixen und dort einen PR stellen zu können .... 
Es ist ja nicht nur das Teilen / 100, auch die Reduktion auf NN fehlt.

Als ich meiner Frau den Sensor im Internet gezeigt hab, war die allererste Frage warum der Luftdruck falsch ist... und es ging nicht um Pa und hPa, sondern warum da 910hPa (bzw. 91000.00 Pa) steht, statt 1023 hPa wie auf unserer Wetterstation ...
Auf Erklärungsversuche kam dann nur ein verständnisloses "Aha ...", was das ganze Projekt ad hoc in die Sparte "Bastellösung" einsortiert hat. Daß wir uns nicht falsch verstehen : sie arbeitet auch in der IT, aber mit nicht vergleichbaren Druckwerten kann halt niemand was anfangen ...

Das ist zwar eigentlich ein anderes Thema (das zugehörige Issue hier ist glaube ich seit 2017 offen und kurz vor den Rejecten) passt aber in diesem Zusammenhang. Auf opensensemap kann man zwar eine GPS Höhe eingeben, aber es passiert dann nichts damit (vllt. sind's auch nur die "fehlerhaften" Pa Werte...). Auch auf Madavi.de oder in der Feinstaub-App wird  nur der angezeigte Druck ausgegeben und nicht reduziert auf NN. Hab dort einen Feature Request aufgemacht, der wurde aber freundlich auf "vielleicht später mal" (aka "nie") verschoben ...

Habe gestern abend noch einen Fork angelegt. Werd das "geteilt-durch-100" einbauen für die Kombination opensensemap und bme280. Das ist allemal weniger Aufwand, als der in meinen Augen etwas gewagte Vorschlag einen Raspi dazwischenzusetzen !
Einserseits bieten wir den Leuten hier ein Produkt als "Ihr müßt nicht Programmieren können", andererseits dann so was ...

Dann schauen wir mal was auf Opensensemap passiert, ob dann die Höhenkorrektur funktioniert.
Und wenn nicht, werde ich mir auch die Reduktion auf NN vornehmen (per Höheneingabe und Flag zu setzen). Mal schauen ob das von der Rechengenauigkeit in der ESP8266 Architektur geht ...

Gruß,
Thomas

ricki-z, 2020-01-27:
Zu Madavi.de: Dort ist die Höhe der Geräte nicht bekannt. Da wird eine Höhenkorrektur schwierig. Die Grafiken laufen ja ohne Registrierung und war eigentlich nur zur Funktionsprüfung der Geräte gedacht.