opendata-stuttgart/sensors-software #74: How to connect a BMP180 sensor?

reibuehl, 2017-05-08:
I have a sensor that currently uses a DHT temp/hum sensor. I would like to upgrade to a BMP180 to also get pressure readings. How do I connect the BMP? Do I need to remove the DHT= Do I need a new Sensor ID for the BMP or will the software use the one of the DHT?

hakan42, 2017-05-08:
I connected a BME280 via I2C, the BMP180 should be the same. The backend seems to use a new Sensor ID.

Also, I left the DHT in the sensor as well so I can compare the readings of the DHT to the ones of the BME280.

From the code:

```
/* Extensions connected via I2C:                                 *
/* BMP180, BME280, OLED Display with SSD1309                     *
/*                                                               *
/* Wiring Instruction                                            *
/* (see labels on display or sensor board)                       *
/*      VCC       ->     Pin 3V3                                 *
/*      GND       ->     Pin GND                                 *
/*      SCL       ->     Pin D4 (GPIO2)                          *
/*      SDA       ->     Pin D3 (GPIO0)                          *
/*                                                               *
```

ricki-z, 2017-05-08:
We need to change the sensor type in the database. Data is sent with different 'pin' per sensor. If you dont remove the DTH22, you will get a new ID. Otherwise we will reuse the DHT ID.

closed 2017-05-24 by ricki-z