opendata-stuttgart/sensors-software #77: platformio not working

ricki-z, 2017-05-11:
platformio with standard espressif8266 compiles with error '.pioenvs/nodemcuv2_bg/firmware.elf section `.text' will not fit in region `iram1_0_seg'
with espressif8266_stage compile is okay, but SPIFFS is not read. Saved config isn't found and connect to internal webserver is not possible.

hakan42, 2017-05-11:
could you please try replacing

```
board = nodemcuv2
```

with

```
board = nodemcu
```

and run the "pio run -t upload" again? If this works, send a message here, I would create a fix pull request then.

darcoto, 2017-05-12:
Unfortunately I still can not reproduce none of both problems - with  section '.text' will not fit in region 'iram1_0_seg' neither with "SPIFFS is not read"

I successfully compile, upload and use latest firmware on Mac and Ubuntu.
I try it dozens of times with espressif8266_stage and with espressif8266 platforms - result is same - working normal.
Also try it with three different hardware versions of esp8266 - no problem again.

Give some ideas ?

@ricki-z I can give you access to my Ubutu machine to compare the eviroment. My be the problem is with some library version.

darcoto, 2017-05-19:
@ricki-z 
Any news for this problem from you ?
Was there a chance to try it again ?

ricki-z, 2017-05-19:
I'm building the firmware versions with Arduino by now. I hadn't time to test it again.
I had a clean install of platformio on an MacOS Sierra. I followed the steps of the 'official' instructions to install the esp8266 extension.

darcoto, 2017-07-06:
@ricki-z I move the problem a little forward 
If I remember correctly, last time you test it, you report that can not save the configuration after compile with platformio.

I reproduce the problem but I have no idea how to fixed.

Here the steps:
- Compile and upload the sketch with platformio
- board boot normal in AP mode
- I connected to it via WiFi, configure SID and password and submit the form
- **Configuration is saved normal** (I see it on serial monitor as JSON)
- Board reboot and connect successful to chosen network and start OTA update
- new firmware successful replace older and reboot
- after reboot **config file missing**

I try this a few time and result was the same - new firmware (build with arduino) delete/overwrite config file (saved with old firmware, build with platformio)

Alternative check - I upload same bin file with arduino->esptool - same result.

Conclusion:
May be there a some internal differences in compile on arduino and platformio and result is - write firmware over config file.

I hope I explained it correctly

Any suggestion from anyone ?

I hope that we can fix it because platformio is a great tool for work on this project

dokape, 2017-09-16:
Does the problem still exists?
@darcoto have you found a solution?

Should we close this issue or keep it open?

closed 2019-01-23 by ricki-z