opendata-stuttgart/sensors-software #79: Sensor interface does not state date of firmware

seb-1204, 2017-05-19:
The sensor interface does not show what date/version the firmware is.

It would be good to have a build date shown in the interface.

ricki-z, 2017-05-19:
Firmware version in shown in the header of all pages. Directly below 'Feinstaubsensor', the ID and the mac address.

seb-1204, 2017-05-20:
Are you referring to 

> Firmware: NRZ-2017-080

Where would I find the correlating tag/version/release information on github?

PS: do the sensors auto-update themselves?

dokape, 2017-09-16:
@seb-1204: 
The sensor-firmware has the auto-update-option. It should be activated.
As there was no additional comments since May, 20:
Are your questions answered? 
Can this issue be closed?

closed 2019-11-27 by dirkmueller