opendata-stuttgart/sensors-software #80: Connection to InfluxDB is broken

YNedderhoff, 2017-05-19:
I have the sensor send data to InfluxDB on my Pi and visualize that in Grafana. From today, early afternoon, there are no data points anymore. InfluxDB is up and running just fine, existing data points are still visible in Grafana, just no new ones are coming in.

Judging from the different looking UI there was a software update recently. Is it possible that something broke so data doesn't get send to Influx correctly anymore?

YNedderhoff, 2017-05-19:
I fixed it. It was so obvious I didn't even check for it. The update overwrote my InfluxDB configuration with the default.

I'll leave it up to you if you think this should be closed or if this is still an issue (as obviously that shouldn't happen).

ricki-z, 2017-05-20:
We renamed some variables. After update these were reset to the default values. I will insert a comment to our sources, that we shouldn't change variable names applied to the API connections.

YNedderhoff, 2017-05-20:
Thanks for looking into it!

ricki-z, 2017-05-21:
So for all InfluxDB and 'Eigene API' users:

SORRY for the mistake in version NRZ-2017-080. We renamed some variables, which were then reset to default. Please change your configuration again to the needed values.

tomru, 2017-06-07:
In case anyone else runs into the issue of having data gaps in your local influxdb (because of this bug, or power outage, etc) feel free to use [these helpers](https://gist.github.com/tomru/4989b51ceef9f1a2ca01aff220800a65) to fill your gaps.

dokape, 2017-09-16:
@YNedderhoff is this issue still actual or can it be closed?

closed 2018-06-09 by ricki-z