opendata-stuttgart/sensors-software #81: Unable to connect to own API on port 443

almsta, 2017-05-20:
Hi!
I want to send the sensor data to my own API on port 443. I'm running an Apache with a proper configured let's encrypt certificate.

The only debug out put via serial is as follows:

```
## Sending to custom api: 
Start connecting to api.xxxxxxxxxx.com
connection failed
```

Is there something "special" I need to configure? Is it possible to get a more detailed debug output?

ricki-z, 2017-05-23:
Please check the configured port for custom API (Eigene API). You can also set the debug level to a higher value there.

almsta, 2017-05-24:
Set the debug level to 5 (debug info high), same debug output as above. Sending data over to port 80 gives me the following output

```
## Sending to custom api: 
Start connecting to api.xxxxxxxxxx.com
Requesting URL: /xxx/sensors/10724178
11111111
{"software_version": "NRZ-2017-080", "sensordatavalues":[{"value_type":"SDS_P1","value":"5.84"},{"value_type":"SDS_P2","value":"5.44"},{"value_type":"temperature","value":"18.20"},{"value_type":"humidity","value":"78.20"},{"value_type":"samples","value":"845661"},{"value_type":"min_micro","value":"164"},{"value_type":"max_micro","value":"499325"},{"value_type":"signal","value":"-78 dBm"}]}

closing connection
------

End connecting to api.grenzstein.com
```

Maybe there is something wrong with my server configuration (Apache as proxy). Is there a way to see which request is made (with full headers etc)? Would be nice for debugging.

ricki-z, 2017-05-24:
Can you try to call the url with a browser?
We can't get the status code with the used libs.

dokape, 2017-09-16:
@almsta is this issue still actual or can it be closed?

almsta, 2017-09-17:
Yes, but I don't have the time currently to dive deeper into it :(

almsta, 2017-11-23:
Finally found the problem. I issued a letsencrypt certificate with a key size of 4096. Seems that the client cannot handle it (memory?). Using a certificate with a key size of 2048 works.

closed 2017-11-23 by almsta