opendata-stuttgart/sensors-software #85: LiquidCrystal_I2C.h: No such file or directory

MarcelScherer, 2017-05-26:
Hallo, 

Hab noch nicht so viel erfahrung mit GitHub projekten. Ich habe mir das Projekt gezogen und alle librarys installiert, aber beim bauen bekomme ich folgende Fehler:
sensors-software-master\airrohr-firmware\airrohr-firmware.ino:76:31: **fatal error: LiquidCrystal_I2C.h**: No such file or directory
Wenn ich die library jm_LiquidCrystal_I2C installiere, fehlt die die nächste library ....

Vieleicht stell ich mich auch zu doof an, aber vieleicht hast du einen Tip für mich.

rexfue, 2017-05-26:
Noch ein wenig weiter runter scrollen bei den Libraries, dann kommt  **LiquidCrystal_I2C**, mit der gehts dann (jedenfalls bei mir).

MarcelScherer, 2017-05-26:
1. 8 Minuten nach der Anfrage schon eine Antwort: "**WOW**", vielen Dank.
Bekomme jetzt noch eine Warnung "WARNING: library LiquidCrystal_I2C claims to run on (avr) architecture(s) and may be incompatible with your current board which runs on (esp8266) architecture(s)."
aber ich hab noch kein board um es zu testen. Vielleicht kann ich ja die warnung ignorieren.

Noch mal vielen DANK

closed 2017-05-26 by MarcelScherer

Uwana, 2019-04-26:
Hello MarcelScherer, how did you resolve the WARNING

MarcelScherer, 2019-04-26:
Sorry but it is two years ago. I'm not shure, but I think, I do not fix it, and I compile it with a warning. But the output works fine.