opendata-stuttgart/sensors-software #86: influxdb ssl support

heinemml, 2017-05-31:
Hi,

I'm trying to feed the sensordata into my influxdb instance but I'm seeing no measurements.

My Influxdb is running on port 8086 and has ssl enabled. From a quick look into the airrohr code I assume that ssl is supported but only used when the port is 443. Wouldn't it be better if I could select "use ssl/tls" explicitly? Is there a specific reason for this or is this just something where there was "no time for implementing it yet".

ricki-z, 2017-05-31:
SSL on esp8266 is very tricky. Any server certificate greater 2048 bytes will crash the node. So we decided not to use SSL for InfluxDB.

heinemml, 2017-05-31:
Thanks for the quick response.

I feared that it might be somewhere in the area of "tls on uControllers". So I guess I'll resort to some kind of proxy solution to get this data to my influx instance.

closed 2017-05-31 by heinemml

darcoto, 2017-05-31:
@heinemml If you familiar a little bit with php I can send you sample of php "proxy" from sensor data to influxdb.

heinemml, 2017-05-31:
@darcoto that would be great

darcoto, 2017-05-31:
This is simple demo that I strip from my code which is more complicated.
Sorry, didn`t test it in this variant, but you must see the idea and twik it 
If you have some question - shoot :)

```php
<?php
$db_name = 'mydb';

$request = json_decode(@file_get_contents('php://input'));
foreach($request->sensordatavalues as $v){
  $values[] = $v->value_type.'='.floatval($v->value);
}

$data = 'dusty,sensor='.$_SERVER['HTTP_X_SENSOR']].' '.implode(' ',$values);

$ch = curl_init();
curl_setopt($ch, CURLOPT_RETURNTRANSFER, true);
curl_setopt($ch, CURLOPT_BINARYTRANSFER, 1);
curl_setopt($ch, CURLOPT_HEADER, true);
curl_setopt($ch, CURLOPT_URL, "https://localhost:8083/write?db=".$db_name);
curl_setopt($ch, CURLOPT_POST, true);
curl_setopt($ch, CURLOPT_POSTFIELDS, $lines);

return curl_exec($ch);```