opendata-stuttgart/sensors-software #87: Meine Version zum Testen evtl Branch ?

jowiemann, 2017-05-31:
Hallo,

anbei meine Version basierend auf der aktuellen "NRZ-2017-085"

Es können folgende Sensoren genutzt werden:
DHT, maximal 3 Sensoren parallel
BMP
BME, zwei Sensoren parallel
TSL2561, drei Sensoren Parallel
VEML6070

Für die Konfiguration der Sensor gibt es eine eigene Konfigurationsseite. Den Sensoren können eigene Namen vergeben werden.

Grüße Jörg

[airrohr-firmware.zip](https://github.com/opendata-stuttgart/sensors-software/files/1042240/airrohr-firmware.zip)

ricki-z, 2017-05-31:
Hallo Jörg,

welchen Sinn haben mehrere Sensoren gleichen Typs? Außer das die Variablen Speicher belegen, der dann die SSL-Verbindungen 'zerstört'. Das gleiche gilt für die dann nötige Konfigurationsseite. Mit deiner Firmware sind 44% des Flashspeichers belegt, nachdem ich gerade wieder auf 41% herunter war. Wir müssen deutlich unter 50% bleiben, da wir sonst keine Auto-Updates machen können. Ich will damit nur verdeutlichen, wieso ich das nicht in die "offizielle" Version übernehme.

Aktuell hängt die Unterstützung der TSL und VEML Sensoren an einem laufenden Betatests, der meine Geräte blockiert. Die Tests laufen jeweils mehrere Tage. Mit einer der nächsten Versionen sollten diese aber Sensoren kommen.

Es steht dir natürlich frei eine eigene Version zu forken.

jowiemann, 2017-05-31:
Hallo Rajko,

Deine Argument kann ich nachvollziehen. Ich habe zwei BME angeschlossen. Einer für die Außentemperatur und einer für die Temperatur im Zimmer. Mein ESP ist innen und die Außen-Sensoren sind über ein Kabel angeschlossen. Da ich die Werte auch selber weiterverarbeite nutze ich zwei TSL um die Helligkeit über den Bereich von Süd-Ost, Süd, und Süd-West abdecken zu können.

Bei der von euch gewählten Vorgehensweise für das automatische Update über OTA habe ich Bauchschmerzen. Mit etwas krimineller Energie können so die ESP gekapert und für unschöne Zwecke missbraucht werden. 

Grüße Jörg

PS: Das ganze ist für mich auch Gerhinjogging, wenn es nicht gebraucht wird habe ich jedenfalls trotzdem die grauen Zellen trainiert.

Von: Rajko Zschiegner
Gesendet: Mittwoch, 31. Mai 2017 19:38
An: opendata-stuttgart/sensors-software
Cc: jowiemann; Author
Betreff: Re: [opendata-stuttgart/sensors-software] Meine Version zum Testenevtl Branch ? (#87)

Hallo Jörg,
welchen Sinn haben mehrere Sensoren gleichen Typs? Außer das die Variablen Speicher belegen, der dann die SSL-Verbindungen 'zerstört'. Das gleiche gilt für die dann nötige Konfigurationsseite. Mit deiner Firmware sind 44% des Flashspeichers belegt, nachdem ich gerade wieder auf 41% herunter war. Wir müssen deutlich unter 50% bleiben, da wir sonst keine Auto-Updates machen können. Ich will damit nur verdeutlichen, wieso ich das nicht in die "offizielle" Version übernehme.
Aktuell hängt die Unterstützung der TSL und VEML Sensoren an einem laufenden Betatests, der meine Geräte blockiert. Die Tests laufen jeweils mehrere Tage. Mit einer der nächsten Versionen sollten diese aber Sensoren kommen.
Es steht dir natürlich frei eine eigene Version zu forken.

dokape, 2017-09-16:
@jowiemann 
Sind deine Fragen soweit beantwortet? 
Siehst Du noch Diskussionsbedarf oder kann diese Diskussion geschlossen werden?

jowiemann, 2017-09-16:
Hallo,

kann geschlossen werden.

Grüße Jörg

Von: dokape
Gesendet: Samstag, 16. September 2017 15:22
An: opendata-stuttgart/sensors-software
Cc: jowiemann; Mention
Betreff: Re: [opendata-stuttgart/sensors-software] Meine Version zum Testenevtl Branch ? (#87)

@jowiemann
Sind deine Fragen soweit beantwortet?
Siehst Du noch Diskussionsbedarf oder kann diese Diskussion geschlossen werden?

closed 2017-09-16 by ricki-z