opendata-stuttgart/sensors-software #88: Improvements: Setup --> Configuration flow

Cinezaster, 2017-06-02:
At Civic labs Belgium, we recently tried to build 10 devices and to be honest the process was a bit hard and inconsistent.
- Flashing 4 of 10 didn't work, could be caused by a long or bad USB cable. We haven't confirmed this yet.
- Connecting to the device in access point mode causes a lot of errors on more then one laptop
- loading the config pages is terribly slow
- The flow of the config pages is not very clear to me and other participants. If you need to reconfigure the sensors (because you don't have DHT22) if you save it and it restarts? If you first fill in SSID and password and save, I don't know a way to go back to the config page.
- The auto update is also causing a bit of confusion and delay. You need to connect over serial and see the serial output, to know this is happening.

Me, I'm using Platformio to configure and flash the ESP8266, and I'm editing the ext_def.h file. This is of course not the easiest route if we want to roll out these devices to none technical people.

# Ideas for improvement
## Flashing
There are a couple of projects like [flasher.js](https://github.com/thingsSDK/flasher.js) a cross platform software tools to flash firmware onto ESP8266 already tried to convert this to push our firmware. Didn't work, but haven't got the time to debug it.
## Better access point performance
I have not enough experience with the ESP8266 to know what we need to improve. I assume the access point/ server code is based on a working example that performs well? 
Next, loading the config page can take up to 30 seconds and often fails. Its this a limitation of the ESP? A solution could be splitting the forms into multiple pages so the amount of data per page is reduced?
Using a jumper to force the device in access-point mode would also be very handy. That way we can keep it in access mode till we're done with configuring the device.
## Config flow
A step by step flow would be preferable because now you have one page with 3 save buttons.
Or we add configuration over the serial port and write a small cross platform app with electron, I did this before and it should be not so difficult. 
## Auto update
How do we know for sure, an auto update wouldn't brick all sensors out there? An LED or something else would be a nice addition to show the status of our device.

ricki-z, 2017-06-03:
## Flashing:
In most cases it's the USB cable. It shouldn't be longer than 50cm (standard USB cable). Otherwise you could try to flash at a lower serial speed.
There is no other tool than the Arduino IDE that is available on mostly all platforms. Any other tools w found had some kind of issues.

## Access point performance:
We could make it faster if we deactivate the search for available WiFi networks. But this is a help for users that are unsure about the name of their WiFi.
Jumpers are just another part of hardware that people need to connect. And so another point of hardware failures.

## Config flow:
There is not enough space on the NodeMCU to build to build a larger amount of pages. Normal users should only need to include the WiFi name and password and click the save button. This is mentioned on the config page ('Change only if ...').
Serial configuration could only be done by people with the suitable knowledge. This is the reason for the configuration via WiFi.

## Auto update:
The updates are tested on more than 4 devices with different configurations. These devices are spread over germany (I need to know someone there, if there is a problem). So there should be only a small chance that devices are bricked by an update.
The devices are mounted outdoor. So an LED is not very helpful to show the status. If you could see the LED you will need to make the LED waterproof. But you can check the sensor via http://feinstaubsensor-<chipID>.local (chipID is the long number we need to include the sensor in our database).

As mentioned in our FAQ, you can reconfigure the sensor with the URL http://feinstaubsensor-<chipID>.local

For flashing it should be possible to find someone who can do this for the people.

And if you test the configuration in a room with many active devices, yes, than there may be problems with wifi connections. As in any room with too many wifi routers around.

Cinezaster, 2017-06-16:
Hi @ricki-z 
Thanks for explaining why and how.

At the moment I have a bit more experience with nodeMcu and flashing and configuring. Have made some bash scripts to help me flashing the devices. 
I'm writing a troubleshoot guide in english which I will share soon.

The only issue that keeps bugging me is connecting to the device in access mode (first time). A few things make this process and flow, slow and unpredictable.

- Connecting my laptop to the nodeMcu in access mode. I have to try more then once before I can connect. Don't know why. No output on serial port. 
- Same for surfing the first time to http://192.168.4.1/ keeps on loading
The solution is terminating the request of the page and refreshing the page and this is what happend at the last 10 lines. The 2 first `output config page` lines did not result in to seeing that page.
```Starting WiFiManager
AP ID: Feinstaubsensor-2434391
output not found page...
validate request auth...
output config page ...
scan for wifi networks...
wifi networks found: 39
output not found page...
output not found page...
validate request auth...
output config page ...
scan for wifi networks...
wifi networks found: 42
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
output not found page...
validate request auth...
output config page ...
scan for wifi networks...
wifi networks found: 43
output luftdaten.info logo...
output codefor.de logo...
output not found page...
output not found page...
validate request auth...
output config page ...
output luftdaten.info logo...
output codefor.de logo...
```
- The user connects to a device with no internet, and the user has a lot of internet hungry applications running searching to connect to there services. All the request are handled by the nodeMcu. Not completely sure but I see a lot of `output not found page....`

I agree with you on showing available networks to the user is a must have. 
We might request a list every minute and store these for use at the config page?

And I agree with you on adding more hardware (leds, jumpers, ...).

Cinezaster, 2017-06-16:
Just saw pull request #93 
I'll test this later.

ricki-z, 2019-11-06:
@Cinezaster Can we close this issue?

Cinezaster, 2019-11-07:
@ricki-z Yes you can.

closed 2019-11-07 by Cinezaster