opendata-stuttgart/sensors-software #89: Deep sleep option

dadosch, 2017-06-03:
The ESP8266 has a so-called deep-sleep function which allows running on a battery much longer. How about adding a option "Nach jedem Messvorgang in den Tiefschlaf gehen, um den Stromverbrauch zu senken" to the webconfig?

With this a solar-power ESP would consume almost _no_ power when not connected to WIFI and not measuring.

A downside would be with the option enabled, that you can't see the live data directly from the sensor.

`ESP.deepSleep(time)`

ricki-z, 2017-06-03:
Much fun, if you try this ;-)
Setting the ESP to deep sleep will wake up the SDS011. Why? The GPIOs will be set to 1 on deep sleep, which will restart the sleeping SDS011.
So you will save some mA for the NodeMCU in exchange for 100 mA needed by the running SDS011.
This could only be solved by a circuit that interrupts the power lines to the SDS011.

dadosch, 2017-06-05:
I remember about someone (on twitter?) with a PCB with an ESP8266 (not NodeMCU) and some LoRa circuitry and proper connectors on it. Do you know what I mean or did I dream this?

dadosch, 2017-06-05:
Found it:
https://twitter.com/guido_burger/status/846689461270446081
https://twitter.com/guido_burger/status/850267793194852352

It was Octopus from http://fab-lab.eu/ which is based on a honeywell pm sensor.

closed 2017-06-05 by dadosch