opendata-stuttgart/sensors-software #91: Can't flash (FLASH_DOWNLOAD_BEGIN failed)

pauledd, 2017-06-04:
Hi there,

I just got my nodemcu (V3, LoLin, black, with FLASH/RESET button) from china.

https://de.aliexpress.com/item/New-Wireless-module-CH340-NodeMcu-V3-Lua-WIFI-Internet-of-Things-development-board-based-ESP8266/32683073762.html

Until the other hardware arrives I just wanted to flash the firmware to that device but I get the same error in linux and windows, tried 3 different cables...

When I upload the blue lamp blinks a little bit but then happens nothing else.

Heres the complete log:
```
~/.arduino15/packages/esp8266/tools/esptool/0.4.9/esptool -vv -cd nodemcu -cb 57600 -ca 0x00000 -cp /dev/ttyUSB0 -cf ~/store/projekte/feinstaub/latest_de.bin 
esptool v0.4.9 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
        setting board to nodemcu
        setting baudrate from 115200 to 57600
        setting address from 0x00000000 to 0x00000000
        setting port from /dev/ttyUSB0 to /dev/ttyUSB0
        espcomm_upload_file
        espcomm_upload_mem
opening port /dev/ttyUSB0 at 57600
        tcgetattr
        tcsetattr
        serial open
opening bootloader
resetting board
trying to connect
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
trying to connect
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
Uploading 439808 bytes from /home/paul/store/projekte/feinstaub/latest_de.bin flash at 0x00000000
        erasing flash
        size: 06b600 address: 000000
        first_sector_index: 0
        total_sector_count: 108
        head_sector_count: 16
        adjusted_sector_count: 92
        erase_size: 05c000
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
        setting timeout 15000
        setting timeout 100
warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
closing bootloader
error: espcomm_upload_mem failed
```

I just read about some kind of "enabling boot mode". I tried to press the FLASH button while issuing the command and I tried both buttons but nothing happens.
 Is there maybe someone having the same trouble or having a solution to that?

ricki-z, 2017-06-04:
Please try to flash again, this time hold the flash button while starting the esptool. Keep the button pressed until the blue LED ist blinking fast for a second. Then you can release the flash button.
Newer versions of the NodeMCU should emulate the flash button in hardware. But this seems to be broken in some cases.

ricki-z, 2017-06-04:
If this won't work the flash memory may be corrupt.

pauledd, 2017-06-04:
Thanks, it does not help, I'll send it back.

pauledd, 2017-06-04:
I found the problem! After inspecting the board very close I found a solder link between PIN SO and SK.
I removed the solder between the pins I was able to flash.

closed 2017-06-04 by pauledd