opendata-stuttgart/sensors-software #96: Feinstaubsensor - Nach zusammenbau keine Daten

DH1OK, 2017-06-18:
Hallo,

ich hab mir einen Feinstaubsensor entsprechend der Anleitung von Luftinfo zusammengebaut. Die Inbetriebnahme des ESP lief wie beschrieben einwandfrei. Nun kann ich den Sensor im WLAN erkennen und auf seine Webseite gehen. Alles ist soweit in Ordnung, bis auf das alle Sensoren keine Werte zurückliefern. Weder der SDS011 noch der DHT22 oder ein inzwischen angeschlossener BME280 zeigt auf der Webseite unter aktuelle Werte irgendetwas an.

Gibt es eine Möglichkeit, eine Fehlersuche durchzugehen? Fehlerlogs auszulesen o. ä? Ich konnte leider hier auf dem Git nichts entsprechendes finden.

Bin über jeden Hinweis dankbar.

Olaf

TheNitek, 2017-06-18:
Du musst dem Sensor ein bißchen (ein paar Minuten) Zeit geben, um die ersten Daten zu erfassen. Warst du evtl. etwas ungeduldig und hast zu schnell resettet?

ricki-z, 2017-06-18:
Es werden erst Daten angezeigt, wenn der Sensor mit einem WLAN verbunden ist. Das hat softwareseitige Gründe. Und es macht wenig Sinn, Daten zu erfassen, wenn diese nicht via WLAN an einen Rechner gesendet werden können.
Wenn eine WLAN-Verbindung besteht, dauert es mindestens 2,5 Minuten, bis die ersten Messwerte zu sehen sind. Wir haben zur Erhöhung der Lebensdauer die Dauer des Messzyklus auf 2,5 Minuten gesetzt. Damit sollte der SDS011 ca. 5 Jahre durchhalten.

DH1OK, 2017-06-18:
Hallo und danke für die schnelle Antwort. Ich hab mir sowas schon gedacht und daher über eine Stunde gewartet.  Leider kam bisher nichts an.

ricki-z, 2017-06-18:
Konfiguration geprüft (Häkchen vor den Sensoren)? Welche Zeilen werden auf der Werte-Seite angezeigt?

ricki-z, 2017-06-18:
Wenn der Sensor am Rechner angeschlossen ist, kann die Debug-Ausgabe des Sensors z.B. über die Serielle Konsole der Arduino IDE angezeigt werden.

dokape, 2017-06-18:
Prüfe nochmals die Verdrahtung.
Check das Serielle Log per Arduino IDE. 
Als Amateurfunker solltest Du ein Multimeter haben: Stimmt die Stromversorgung an den Sensoren?
Kommt 5V /3,3V dort an? 
Hast Du ein Oszi? Gibt es auf den Datenleitungen RX/TX Datenverkehr? 
Hast Du ein anderes Netzteil probiert? 
Ein anderes Micro-USB-Kabel?

DH1OK, 2017-06-18:
Hallo
Ich werd mich morgen Abend nochmal ran setzen. Mal sehen was die serielle Ausgabe so macht. Spannungen hab ich gemessen sind i.o. Tx/Rx muss ich mal sehen aber Danke für den Ansatz.

Ich werde die Ergebnisse mitteilen. Danke

DH1OK, 2017-06-19:
Hallo

also das ist jetzt komisch. Der Sensor hat Zugang zum WLAN und eine IP-Adresse von der Fritz-Box bekommen. Die Oberfläche des Sensors ist erreichbar über die vergebene IP-Adresse der FritzBox. Ich kann alle Schaltflächen anklicken.

Wenn ich nun auf die Schaltfläche "Aktuelle Werte" klicke, erhalte ich zwar die Zeilen für SDS 011 DHT 22 und Wifi zurück, aber keine Werte. Nur bei Wifi steht -50DBm und 100% Qualität. 
Ändere ich die Konfiguration, dann erscheinen mehr Zeilen oder auch weniger, je nachdem was ich verändere. Ich hab z.B. einen BMP280 zusätzlich oder auch einmal nur den DHT22 in der Konfiguration angehakt. Aber Werte hab ich nie erhalten.

Spannungen hab ich alle überprüft, die sind i.O. (4,87 und 3,32 Volt) Ebenso sehe ich den Datenverkehr auf den Leitungen TX und RX. Der Sensor scheint mir Daten auszugeben. Verdrahtung hab ich auch schon mehrmals geprüft und getauscht. Die Hardware steht bei mir auf dem Tisch auf einem Breadboard zusammengesteckt, da lässt sich die Verdrahtung schnell ändern.

Ich habe den ESP an den PC angeschlossen und die serielle Schnittstelle zum richtigen Com (wie bei der Firmware) eingestellt. Im Arduino IDE hab ich die serielle Schnittstelle eingeschaltet zum mitlesen. Egal welchen Debug-Level ich über dei Schaltflächen der Web-Oberfläche anschalte , auf der Schnittstelle kommt nichts raus. Einzig wenn ich auf der Weboberfläche die Schatfläche "Sensor neu starten" anklicke, dann sehe ich folgende Ausgabe :

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld

Mehr nicht. 

Kann ich den ESP irgendwie wieder in den "leeren" Zusatand zurücksetzen? Die Firmware neu einspielen hab ich schon gemacht, hat aber bei mir nichts geändert.

Olaf

DH1OK, 2017-06-19:
Firmware ist die 
NRZ-2017-090
Das scheint die aktuellste Version zu sein, oder?

ricki-z, 2017-06-19:
Auch dort gilt: Erste Messung nach 2,5 Minuten.

ricki-z, 2017-06-19:
Wenn dort nach dieser Zeit keine Daten stehen, wird zwar versucht, diese zu lesen, aber es kann nichts gelesen werden. Was dann für falsch angeschlossene Sensoren spricht.
Über 90% Prozent der Sensoren laufen mit dieser Software-Version, und das bisher ohne Probleme (zumindest softwareseitig).

ricki-z, 2017-06-19:
Und ja, wenn man die Sensoren ändert, ändert sich auch die Werteseite. Warum sollte sie das auch nicht tun? Wenn der Benutzer angibt, daß es einen BME280 gibt, sollte die Software davon ausgehen können, das dem auch so ist.

DH1OK, 2017-06-19:
Der Sensor läuft inzwischen seit mehreren Stunden. Keinerlei Werte, egal was ich konfiguriere. Die Softwareseite scheint zu funktionieren, ebenso die Seite der externen Sensoren. Ichhabe jedoch die GPIO's in Verdacht. Ich werde mal einen neuen ESP ordern und berichte dann wieder.

dokape, 2017-06-19:
Wenn Du die GPIO im Verdacht hast: 
Da gibt hier im Repo eine Testprogramm für die GPIO, hatte ich kürzlich gesehn.
Evlt. kannste damit die Pins mal testen? 
Habs selbst nicht benutzt, nur beim stöbern gefunden mal Code kurz überflogen.

https://github.com/opendata-stuttgart/sensors-software/tree/master/esp8266-arduino/pintester

DH1OK, 2017-06-19:
Danke ein Ansatz für morgen!

DH1OK, 2017-06-21:
Also was soll ich sagen. Aus Fehlern? lernt man? Ich hatte einen ESP8266-12E V3 bei Am...n geordert. Bekommen habe ich ein Board, mit der Aufschrift LoLinNodeMCU V3 . Bei diesem Board scheinen mir die Pins nicht zu stimmen. Ich kann alles mögliche programmieren, aber die Verbindung nach Außen klappt irgendwie nicht. Weder i2c noch LED blinken :-(
Heute kam die Post mir einer weiteren ESP8266-12E vorbei. Es stellte sich heraus, dass es sich um eine V2 Version handelte. Also noch kein VU auf dem Board. Alle zuvor geschriebenen Sketche liefen mit diesem Board einwandfrei. Und nun auch der Feinstaubsensor. Er liefert Werte zurück, so wie es sein soll.
Ich werde jetzt noch ein bisschen experimentieren müssen und die einzelnen Boards mal untersuchen. Vielleich komm ich ja drauf, was das Problem ist.

Danke an alle

closed 2017-06-21 by DH1OK

dokape, 2017-06-21:
Interessant. Hier habe ich auch noch eine Lolin nodeMCU V3 von wemos.cc (über Aliexpress)
Geflashed ist sie bereits, aber noch keine Sensoren dran gewesen. Das mach ich nun gleich mal.

ricki-z, 2017-06-21:
Zu Version 1 und 2 der NodeMCU:
Für die Versionen 1 und 2 sind als VU und G die Anschlüsse VIN und GND neben dem MicroUSB-Anschluß zu nutzen.