opendata-stuttgart/sensors-software #98: license violation

jonasstein, 2017-07-02:
please check the file
https://github.com/opendata-stuttgart/sensors-software/blob/master/airrohr-firmware/pdf/sds_dht_wiring.pdf
it uses non free material.

ricki-z, 2017-07-02:
PDF ist removed. It was a first draft for a documentation.

dokape, 2017-07-21:
Which part was not licenced? 
The images of DHT and/or SDS011? 
Or the pinout of the nodeMCU?
In Case of pinout nodeMCU the wiki should be checked to:

https://github.com/opendata-stuttgart/meta/wiki/Pinouts-NodeMCU-v2,-v3

In case of image for DHT and SDS I can make some new images.

ricki-z, 2017-07-21:
The mentioned pdf was removed. It was was first draft for our 'manual' and someone used images from aliexpress. We don't need this pdf anymore.

dokape, 2017-09-11:
Can this issue be closed?

jonasstein, 2017-09-11:
yes, the file was removed. I close it for the developers here.

closed 2017-09-11 by jonasstein