opendata-stuttgart/sensors-software #99: Extension: Solar Panel + Battery

ccoenen, 2017-07-03:
I'm currently planning on building a few of these sensors with friends. The question came up how to supply those things with enough power without wiring them up.

Can anyone tell me, how much energy a complete setup requires constantly and how much it consumes peak?

ricki-z, 2017-07-03:
The complete setup requires 40 to 50 mA, if the SDS011 is inactive. And 150 to 160 mA if the SDS011 is active. This is the case for 20 seconds per 150 seconds. So the average consumtion should be around 65 mA.

ccoenen, 2017-07-03:
That's amazing! Then two 18650 lithium batteries or typical smartphone batteries should easily drive it for over 24h. And a 5V/1W Solar panel might be enough to drive it _and_ recharge the battery :-D

[This is about twice as powerful](https://www.conrad.de/de/duennschicht-solarmodul-2-wp-6-v-amorphes-110605.html), costs ~15€, and is not _too_ large at 30x16cm:

ricki-z, 2017-07-03:
For the winter month you should calculate some more needed solar power. There are fewer hours with sun and more where you need to run on battery power.

dokape, 2017-07-04:
A powerbank for smartphones with 20 000mAh can drive the Sensor at about 2,5 days.

Cinezaster, 2017-07-07:
My sensor stopped working after 58h 51m (2,4 days) connected to an older 11.200mAh battery.
@dokape not all powerbanks are up to spec. A 20.000mAh should be capable of running 4,3 days, compared to my test.

ccoenen, 2017-07-07:
Both of these seem quite short IMO.
11.200mAh / 58h -> 193mA
20.000mAh / 60h -> 333mA
Granted, there is probably a lot of voltage conversion involved. Maybe this could be optimized.

Both of you sound, as if you were driving your devices off of the 5V USB power, which means that some voltage regulator creates 5V just so you can step down to 3.3v a little later for the ESP chip. That's probably a 10-20% loss each time. Perhaps the ESP could be driven directly by the battery (with protection but without step-up-regulator).

40mA while idle also sounds like a lot. The ESP has a few sleep modes which might bring it down further. In deep sleep, the ESP itself shouldn't require more than a few Milliampere.

ricki-z, 2017-07-07:
The idle modes need some more 'hardware'. If the ESP8266 goes to deep sleep, the GPIOs are set to high. This will wake up the SDS011. So for saving some of the 40mA you will get 100mA more consumption.
To solve this you need a setup that is deactivating the power supply for the SDS011 in deep sleep.

dokape, 2017-07-10:
Yes, @Cinezaster @ccoenen , i know the specs of the powerbanks are sometimes unreasonable. 
So it was just my experience with two same better quality powerbanks the first weeks of sensorusage until powerline war installed.

So the batterie should be able to provide 48hours 200mA to run the sensor when weather is bad.
The solarcell should be able to refill the batterie within a few sunny hours or in winter to fill it in 6 cloudy hours. 
I would appreciate a 20 to 25Wp Solar Module and a 20 aH Batterie. That should be able to run the sensor even an a few cloudy days in winter. In Sommer a 5 to 10WP solar Panel could be enough. 
(I did not made any calculation, just going from the experience above. )
By usage of Powerbank you need a bank that can be loaded during usage. 
But it could be besser to use a 12V Batterie for a 12V Solar-Module and then 
use a DC-DC- Changer 12 to 5V with less drop.
All together would be around 100Euros for the parts, i guess. 
Any suggestions?

Cinezaster, 2017-07-11:
I tested again with the 11200mAh battery. Got 78,3 hours out of it.
Next test will be with an 6W solar panel connected to charge port of battery pack. Although the test is going to be a bit worthless because amount of sunlight is hard to measure.

ricki-z, 2017-07-12:
Some of the next sensors will be a light and an uv sensor. So it should then be possible to measure sunlight ;-)

ccoenen, 2017-07-24:
(just leaving this here for further reference http://www.instructables.com/id/ESP32-Solar-Weather-Station/ )

Gitsaibot, 2017-08-28:
Ich spiele auch mit dem Gedanken den Sensor mit Solar Panel und Batterie zu betreiben. Was mich etwas stutzig macht ist die Diskrepanz zwischen den oben erwähnten durchschnittlichen 65mA und den unten errechneten 193mA; 143mA. Selbst wenn die Angaben auf den Akkus nicht ganz stimmen sind das mehr als das Doppelte. Im FAQ ist von einem Watt die Rede werden da die Verluste vom Netzteil mit eingerechnet!? Leider sind meine Komponenten noch auf dem Weg aus China hierher sodass ich noch nicht selbst messen konnte. 

Für meinen Solar Testaufbau dachte ich an sowas: kebidu 2016 Portable Dual USB Solar Panel Battery Charger 5V 3.6W 500mA

Ich bin mir nun aber nicht mehr sicher ob das wirklich ausreicht um über den Winter zu kommen ? Etwas Reserve soll ja auch noch bleiben falls weitere Sensoren dazukommen. Hat hier jemand schon eine stabile Kombination aus Panel und Batterie am laufen ? 

Ich hoffe ihr verzeiht mir wenn ich auf deutsch schreibe aber so geht das für mich schneller und da hier ja eh alle der Sprache mächtig sind sollte das passen ;).

ricki-z, 2017-08-28:
Die 193mA und 143mA sind errechnet aus der Laufzeit des Sensors an den genannten Powerbanks. Leider liefern diese nie die komplette Energie. Wenn man also ein schlechtes 20.000-mAh-Exemplar erwischt, kommt vielleicht nur die Hälfte raus und die errechnete Stromaufnahme verdoppelt sich.

Adorfer, 2017-08-28:
Mit diesem Ansatz werdet ihr wirklich nicht weiterkommen. 
Siehe auch 
https://github.com/opendata-stuttgart/sensors-software/issues/104

Gitsaibot, 2017-08-29:
@Adorfer Danke für die Links zu den Videos. Das erspart mir schon mal viele Tests ;). Warum man mit diesem Ansatz nicht weiterkommen soll verstehe ich nicht ganz ?

Im Grunde sind folgende Dinge zu klären:

1. Fläche für die Solarpanel(wird wohl größer werden als gedacht)
2. Akku + Ladeelektronik wobei wohl die Powerbanks teilweise schon eine Ladeelektronik haben. 
3. Ist man bereit für eine Solarlösung soviel Geld auszugeben

Da der Sensor selbst nur ~30€ kostet und eine Solarlösung wohl unter 40€ nicht machbar wäre ist Punkt 3 für mich der Entscheidende, da ich ihn an dem geplanten Aufstellungsort auch über die Steckdose versorgen könnte. Amortisieren wird sich das bei 2,50€ Betriebskosten wohl nie. Vorher hat "Schnüffel Sensor" seine Betriebsstunden erreicht. 

Ich stimmte Adorfer zu das man wohl für eine reine Solarlösung eine etwas andere Konfiguration des Sensors verwenden sollte(erhöhen der Messintervalle usw.) um die Leistungsaufnahme zu verringern.

Adorfer, 2017-08-29:
Es wird halt nicht funktionieren, die derzeitige Firmware/Hardware mit einer angedocktne Solar-Lösung zu versorgen. Das wird mindestens(!) 1qm Solarpanel benötigen für einen ganzjährigen Betrieb, vermutlich sogar eher 2-3qm. 
(Und ja, das wäre meines Erachtens Irrsinn, allein aus Montage-Gesichtspunkten, weil das ja auch irgendwie wetter- insbesondere windlastfest sein muss am Ende. Und wenn man's straßennah installiert könnte sich auch noch das Vandalismus/Diebstahlproblem ergeben.)

Und wenn man Abstriche macht, dass z.B. im verregneten Januar keine Werte benötigt werden im Berufsverkehr morgens knapp um Sonnenaufgang... eher kontraproduktiv, weil Wintersmog eines der Probleme ist, was man monitoren möchte.

dokape, 2017-09-03:
eine interessante Neuankündigung von Anker-Powerbanks mit Passthrough:

https://www.golem.de/news/anker-powerbank-mit-usb-c-leichtere-akkus-mit-power-delivery-und-passthrough-1709-129850.html

Laden und Gerät versorgen gleichzeitig. Fehlt nur noch die passend Dimensionierte Solarzelle.

Adorfer, 2017-09-05:
Vielleicht fehlt mir gerade das Verstädnis, was ist an dem Ding jetzt besonders?
Ausser vielleicht, dass die Anker-Fanboys jubeln können "Anker ist aufgewacht und baut jetzt PassThrough ein, wo RAVpower und andere das schon seit vielen Jahren können. (Und nein, wir brauchen kein USB-C hier. Allein die Doppelwandlung, die für "erstmal auf 5V" ist grober Unfug, der nur gerechtfertig wäre, wenn das Problem "Solarzellen alle zu groß" wäre. 
(Bitte lese doch die vorangegangenen Postings und versuche es zu verstehen. ohne MPT-Wandler wird's im November oder Januar wirklich nichts.)

bbwonder, 2018-01-21:
I've been struggling with winter sun & have ordered one of these:
https://rover.ebay.com/rover/0/0/0?mpre=https%3A%2F%2Fwww.ebay.co.uk%2Fulk%2Fitm%2F302352597004
I will report back with results!
I plan to cycle power 60s on, 120s off - hoping to save ~60% energy requirements.

martin-zh, 2019-06-17:
I was looking into the possibiliy of running the sensor by using solar power. I found this Project here:
https://community.hiveeyes.org/t/versuchsaufbau-autonome-zelle-solar-feinstaub-wetter-vergleichsding/1373/10

Is anbody still working on a PV solution?

DeeKey, 2019-12-30:
My opinion is that ESP8266 is not a good thing to use for such a setup. It is much better to use ESP32. It has battery support and also uses less energy.

Anyway, @ricki-z please put FeatureRequest tag to mark that issue properly.