opendata-stuttgart/sensors-software #100: Undocumented board: Adafruit feather M0 Lora

Cinezaster, 2017-07-05:
In the firmware we see the implementation of the adafruit feather M0 Lora board.
- Is this working
- How does it work
- To which network does it connect
- How does the data get to the map
- ??

ricki-z, 2017-07-05:
This were the first steps to get a working version for "TheThingsNetwork". But the development is actually stopped. We don't have the needed infrastructure for this in Stuttgart.

Cinezaster, 2017-07-05:
@ricki-z Thx.
I was already doing some test with the RFM95W and the ESP8266 and the LMIC stack, but no success yet. I do need more time for this.
Is there a MQTT broker if we manage to get it working?

rexfue, 2017-07-05:
@Cinezaster
What are your problems ? I do have a running node with excatly the same configuration. May be, 
I can help?

Reinhard

Cinezaster, 2017-07-06:
@rexfue, I found [your code for the LoPy](https://github.com/rexfue/Feinstaub_LoPy), Thanks for the tip, I'll order one and start testing. Although Python is not a language I master yet. 

Did you manage to have LoRa working on an ESP8266 NodeMcu with a RFM95W as wel? Because I have a couple of those laying around. Would be nice to find some example code and a connection diagram. 

Our goal is to have our sensors next to official sensors but we don't have wifi or power at those location, we could add a mobile phone and tether the data connection over wifi. But a LoRa solution would be easier considering it uses less power and doesn't require Wifi access.

rexfue, 2017-07-06:
@Cinezaster
> Did you manage to have LoRa working on an ESP8266 NodeMcu with a RFM95W as wel? Because I have a couple of those laying around. Would be nice to find some example code and a connection diagram.

Yes I do. I will put description and software on github during coming weekend.

Cinezaster, 2017-07-06:
@rexfue You're a genius, I and all the members of civic labs Belgium are going to be very happy with your effort.

rexfue, 2017-07-08:
@cinezaster  Software and description are on github ( rexfue/Feinstaub_LoRa_ESP8266 )
I hope, I haven*t omitted anything. If there are questions, feel free to mail me  ( rexfue@gmail.com <mailto:rexfue@gmail.com> ).

Cinezaster, 2017-07-20:
After successfully running the Lora version of @rexfue I think Lora should not be a part of this firmware.

I'm going to improve the usability of the Lora script in a [fork](https://github.com/CivicLabsBelgium/Feinstaub_LoRa_ESP8266) of @rexfue [firmware](https://github.com/rexfue/Feinstaub_LoRa_ESP8266)  
Since we're going to develop a custom pcb combining ESP8266, RFM95W, SDS011 and DHT22 we're also looking into making this sensor battery powered and adding the necessary hardware to actually go into deep sleep.

closed 2017-07-20 by Cinezaster

ErwinHartogsveld, 2019-07-09:
Here also a working LoRa combination.
https://github.com/ErwinHartogsveld/Feather_32U4_SDS011_BME280_TTN