opendata-stuttgart/sensors-software #103: Request: Support for PMS7003

Adorfer, 2017-07-14:
http://aqicn.org/sensor/pms5003-7003/de/ seems to support 10, 2.5 and 1.0 . 
In plus it's not only smaller, but promises as PMS7003 higher MTBF due to optimized airflow avoiding dusting. 

[image]
[image]

(perhaps it is already a drop-in, but the extra 1.0 scale -in case it works- will probably need some adaptation.)

Adorfer, 2017-07-14:
sample code see 
https://github.com/MartyMacGyver/PMS7003-on-Particle/blob/master/pms7003-photon-demo-1/pms7003-photon-demo-1.ino

ricki-z, 2017-08-17:
New version NarZ-2017-095 with support for Plantower Sensors is online. Sensors should get this Version via auto update. Please check this version and close this issue if it's okay. PMS7003 is connected the same way as the SDS011.

ricki-z, 2017-09-15:
@Adorfer actual firmware version is supporting all Plantower dust sensors.
Can we close this issue?

closed 2017-09-16 by Adorfer

Adorfer, 2017-09-21:
Just for note: works now fine: 
[image]

dokape, 2017-10-06:
@Adorfer mein PMS7003 zum testen ist heute gekommen.
Du hast auf den Bildern ein Kabel mit Platine von 2,54mm Raster auf das 1,27mm Raster vom PMS-Stecker.
War das bei dir dabei? Bei mir ist nur ein "Stecker" zum Auflöten auf eine Platine dabei.

Adorfer, 2017-10-06:
Bei mir waren sowohl besagte Adapterplatine mit den Kabeln (war ein Stecker dran, habe ich abgeschnitten, um auf den Wemos-Mini zu löten, zum Stecken passt die Reihenfolge nicht ;-)

Und es war auch dieser winzlings-Stecker (siehe eingekreist im Bild) dabei. 
Was der Lieferant mitliefert: ist je nach Anbieter verschieden. Einzeln sind die Adapter-Platinen jedoch unverschämt teuer. (5 USD?)
[image]

Adorfer, 2017-10-06:
P.S: https://www.aliexpress.com/item/The-G7-switch-board-with-cable-for-laser-sensor-PMS7003-PM2-5-particles/32780006665.html