opendata-stuttgart/sensors-software #105: Keine Messwerte

Kettenknecht, 2017-07-16:
Hallo,

ich habe mir den Feinstaubsensor von Luftdaten Info gebaut. Verbaut habe ich:

E202 New Wireless module CH340 NodeMcu V3 Lua WIFI Internet of Things development board based ESP8266

BME280 Digital Sensor Temperature Humidity Barometric Pressure Sensor Module GY-BME280 I2C SPI 1.8-5V

NOVA PM2.5 Air particle/dust sensor SDS021, laser inside, digital output SDS021 Laser PM2.5 sensor Ultra small particle sensor

Angeschlossen hab ich den SDS021 wie den SDS011. Den BME280 so wie in der FAQ beschrieben. Das Ding bootet und meldet sich auch erfolgreich am Netz an, ich kann über den Browser zugreifen. Der Lüfter des SDS021 läuft auch an. Den SDS021 hab ich auf der Konfigurationsseite als SDS011 angegeben, weil das die einzige Möglichkeit ist.

Das Problem ist, ich bekomme bei "aktuelle Werte" für beide Sensoren 0 angezeigt. Und ich hab nicht die geringste Ahnung wo ich mit der Suche anfangen soll. Man kann zwar ein Debuglog einschalten, aber, wie komm ich an das Log heran? 

Ich hab auch versucht mit "an eigene API" senden erstmal herauszufinden ob da evtl. was sinnvolles kommt, allerdings schlägt auf dem konfigurierten Rechner nix ein. Den Port hab ich mit netcat aufgemacht, wenn ich da von nem anderen Rechner aus was reinkippe kommts da auch raus.  Allerdings bekomm ich weder über tcp noch udp was vom Sensor. 

Viele Grüße, und dankbar für jeden Tip
Steve

ricki-z, 2017-07-16:
Der Debug Output kann über den USB-Anschluß ausgelesen werden. Am einfachsten, wenn man die "Serielle Konsole" der Arduino IDE nutzt.
Wie lang lief der Sensor schon, als du die Werte-Seite geöffnet hast. Die erste Messung läuft erst nach ca. 3 Minuten.

Kettenknecht, 2017-07-16:
Danke, das hilft schon mal weiter. Dass es bis zur ersten Messung etwas dauert zeigt er ja auch an. Aber ich hab den mehrmals ca. ne halbe Stunde laufen lassen. Halt alle Kabel nochmal getauscht, die Firmware nochmal aufgespielt und all die Sachen die man macht wenn man keine Ahnung hat wo ansetzen, aber erstmal eigene Fehler ausschliessen will.

Kettenknecht, 2017-07-16:
Zum einen scheint mir der BME hinüber zu sein, die Kabel sind alle ordentlich dran und haben auch Durchgang. Was er dann mit dem SDS macht erschliesst sich mir nicht ganz. Auf jeden Fall scheint er ihn erreichen zu können. Liegts evtl. dran dass das ein SDS021 statt 011 ist?

⸮mounting FS...
mounted file system...
reading config file...
opened config file...
parsed json...
Starte Webserver... 0.0.0.0
6
Connecting to smo.bnfh
.........
WiFi connected
IP address: 10.0.0.127
Starting OTA update ...
NodeMCU firmware : NRZ-2017-092
www.madavi.de
/sensor/update/firmware.php
Start reading SDS011 version date
End reading SDS011 version date
[update] No Update.

ChipId: 585180
Lese SDS...
Lese BME280...
Sende an custom API...
Auto-Update wird ausgeführt...
Trying BME280 sensor on 76 ... not found
Trying BME280 sensor on 77 ... not found
Check BME280 wiring
Stoppe SDS011...
Call sensorSDS
Start reading SDS011
End reading SDS011
Call sensorSDS
Start reading SDS011
End reading SDS011
Call sensorSDS
Start reading SDS011
und so geht das weiter...

Zum Zeitpunkt der ersten Messung versucht er dann Daten zu senden, was aber auch failed, ich hab keine Ahnung warum er keine Verbindung bekommt, es kommt auch nix auf dem konfigurierten Port an. Mehr ist auch nicht.

------
## Sending to custom api: 
Start connecting to 10.0.0.13
connection failed
Time for sending data: 10221
Call sensorSDS

IngoOWL, 2017-07-16:
Wie hast du den BME angeschlossen? 5V oder 3,3V Versorgungspannung? Ich hatte meinen aus Versehen mit 5V betrieben, dann bekommst du auch 5V Datenpegel auf die ESP Ports - nicht wirklich gut, obwohl das bei einen anderen ESP einige Tage gut ging. Bei dem für den Feinstaubsensor hab ich die Ports gegrillt.

Kettenknecht, 2017-07-16:
Angeschlossen ists richtig. Den BME auf 3V, und der SDS auf 5V. Nachgemessen ob das wirklich da rauskommt hab ich allerdings nicht.

dokape, 2017-07-17:
Ich würde mal reduzieren:
Die Funktion vom SDS021 ist ungeklärt -> mal abklemmen. Und in der config abwählen.
In der config nur den den BME280 aktiviert lassen.
Weiterhin auch das Senden an die APIs abwählen. 
Oben die APIs Luftdaten sowie Madavi und unten auch bei den weiteren API alle abwählen.
Dann kann man das Messinterval auf 20sec hochsetzen (musst nicht so lang warten)
Auf der internen Webseite des Sensors http://SENSOR-IP/values 
erscheint zuerst der Countdown und dann die Werte vom BME. 
Teste mal dieses einfache Setup. 
Kontrolliere beim BME nochmals genau die beiden Datenleitungen. Sind die vertauscht?
Hast Du am BME auch die richtigen beiden SDA/SCL erwicht und nicht einen der anderen beiden?
Ist evtl eine Lötbrücke zwischen SCA/CSB/SDO beim einlöten der Stiftleiste entstanden?
Mein BME lief auf Anhieb. Die Temperatur meines BME ist aber um 2-3Grad höher als beim DHL22, der realistischere Werte liefert. 
Hier würde ich - auch im Hinblick auf Vergleichbarkeit mit anderen Airrohren (Sensoren) die vorgesehenen
SDS011 und DHT22 anschließen. 
Ich habe den BME anfangs nur deshalb dran, weil die vorgesehene Bestellung DHT22 nicht ankam. 

Die eigene API nutze ich übrigens so:
Diese Simple-Data-PHP:
https://github.com/opendata-stuttgart/madavi-api/blob/master/data_simple.php
Auf dem netzwerkinternen Raspi im /var/www/html abgelegt und rufe die per http auf. Im Unterverzeichnis /data werden dann für jeden  Tag .csv-Dateien mit den Sensordaten abgelegt.

ricki-z, 2017-07-17:
Die iNova-Sensoren SDS011, SDS018 und SDS021 nutzen alle das gleiche Protokoll. Lediglich die Messwerte unterscheiden sich leicht.
Der SDS018 ist eher als 'Indoor'-Sensor gedacht. Der SDS021 ist eingestellt worden, da die Serien-Abweichungen zu groß waren.

Kettenknecht, 2017-07-28:
Danke für alle Hilfe. Der BME ist wahrscheinlich wirklich hinüber, ich hab noch nen DHT22 bestellt und den angeschlossen, nun funktionierts. Beim SDS021 waren die Datenleitungen tatsächlich falsch herum angeschlossen. Also mein Fehler.

ricki-z, 2017-08-13:
@Kettenknecht : Kann der Issue geschlossen werden?

Kettenknecht, 2017-08-13:
Ja, geht alles einwandfrei. Hätte ich das machen müssen?

Viele Grüße
Steve

closed 2017-08-13 by Kettenknecht

ricki-z, 2017-08-13:
Müssen nicht. Aber es hält die Liste der Issues kurz, wenn diese nach Erledigung geschlossen werden. Sonst sucht später jemand nach einem Fehler, der gar nicht mehr existiert.