opendata-stuttgart/sensors-software #106: 4294967 Sekunden bis zur Messung

dokape, 2017-07-21:
Witziger Bug 
Ist die Messung gerade abgeschlossen und man lädt die Seite mit den Sensorwerten genau in dem Moment direkt von der /values - Seite des Sensors, wird ein vollkommen wirrer Wert ausgegeben. 
Beim nächsten reload der Seite ist alles OK. 

[image]

ricki-z, 2017-07-21:
Ich habe eine dunkle Ahnung, woher das kommt. Ausgegeben wird:
round((sending_intervall_ms - (act_milli - starttime)) / 1000.0)
sending_intervall_ms ist konstant, act_milli wird jeweils zum Beginn eines Schleifendurchlaufs auf die aktuelle 'Zeit' gesetzt. starttime wird einmal in setup() auf die aktuelle 'Zeit' gesetzt, dann jeweils direkt nach dem Senden der Daten.
Und danach erfolgt noch einmal ein Abarbeiten der Page requests. Ich verschiebe jetzt mal das zweite Setzen von starttime hinter die page requests. Dann sollte es eigentlich nicht mehr vorkommen. Kann aber dauern, bis das auf allen Sensoren ankommt ;-)

dokape, 2017-07-21:
Oder dirty sowas:
if Roundwert > Messintervall, dann Roundwert = 0

Fehler ist reproduzierbar, wenn man genau in der abgelaufenen Sekunde ein page-reload macht :-) 
Aber wer macht das schon. 

Die Messwerte liegen ja vor, die erste Messung ist komplett durch. 
Man könnte dann auch einfach die Anzeige der roten Schrift unterdrücken.
If Messung = fertig, dann nix Wartehinweis. 

If Not Messwert, dann Wartehinweis.
Wenn das erste mal Messwerte vorliegen, dann gibts auch keinen Wartehinweis mehr. Dann kann da drinstehen was will.

Wer mal Software-QS gemacht hat, findet überall alle Fehler. ;-)

ricki-z, 2017-07-21:
Das mit If not Messwert war ja drin. Es gibt ein Flag, das nach der ersten Messung gesetzt wird. Leider war aber auch hier die Verarbeitung der Page requests an einer ungünstigen Stelle.

dokape, 2017-09-04:
[image]

Die Änderung vom 21.7. war nicht hilfreich.

dokape, 2017-09-11:
NRZ-2017-099
Tritt nicht mehr auf. Steht jetzt sauber da: 
0 Sekunden bis zur ersten Messung.
Gefolgt (nach reload) von
0 Sekunden seit der letzten Messung und Display der Werte. 

*closed*

closed 2017-09-11 by dokape