opendata-stuttgart/sensors-software #109: Trying BME280 sensor on ... not found

PaulchenPlump, 2017-07-23:
Hallo,

ich habe den Feinstaubsensor erfolgreich mit SDS011 und DHT22 in Betrieb nehmen können und er liefert auch Daten. Allerdings scheitere ich am BME280. :( Egal was ich versuche ich erhalte keine Daten.

Die Kabel habe ich testweise ausgetauscht, genau wie den Sensor (hatte zwei bestellt). Bin mir auch relativ sicher, dass der Anschluss wie im Wiki beschrieben erfolgt ist... Auf der config-Seite habe ich den Sensor aktiviert, aber nach einen Neustart ist der Haken wieder raus. :(

Muss ich noch etwas Software-seitig aktivieren?

Hier mal das Debug-Log nach dem Start:
.........
WiFi connected
IP address: xxx.xxx.xxx.xxx
Starting OTA update ...
NodeMCU firmware : NRZ-2017-092
www.madavi.de
/sensor/update/firmware.php
Start reading SDS011 version date
0 - 173 - ad - 173 .
0 - 40 - 28 - 40 .
0 - 16 - 10 - 16 .
0 - 250 - fa - 250 .
0 - 170 - aa - 170 .
1 - 197 - c5 - 197 .
2 - 7 - 7 - 7 .
3 - 17 - 11 - 17 .
4 - 4 - 4 - 4 .
5 - 18 - 12 - 18 .
6 - 29 - 1d - 29 .
7 - 239 - ef - 239 .
8 - 58 - 3a - 58 .
Checksum is: 58 - should: 58
9 - 171 - ab - 171 .
SDS version date : 17-4-18
SDS device ID:     1def
End reading SDS011 version date
[update] No Update.

ChipId: 2525376
Lese SDS...
Lese DHT...
Lese BME280...
Auto-Update wird ausgeführt...
Trying BME280 sensor on 76 ... not found
Trying BME280 sensor on 77 ... not found
Check BME280 wiring
Stoppe SDS011...
Call sensorSDS
Start reading SDS011
End reading SDS011

Der DHT22 ist zur Zeit nicht angeschlossen, aber aktivert - hier mal ein Log der Sensordaten:
Start reading SDS011
0 - 170 - aa - 170 .
1 - 192 - c0 - 192 .
2 - 18 - 12 - 18 .
3 - 0 - 0 - 0 .
4 - 89 - 59 - 89 .
5 - 0 - 0 - 0 .
6 - 29 - 1d - 29 .
7 - 239 - ef - 239 .
8 - 119 - 77 - 119 .
Checksum is: 119 - should: 119
9 - 171 - ab - 171 .
PM10 (sec.) : 8.90
PM2.5 (sec.): 1.80
End reading SDS011
Call sensorSDS
Start reading SDS011
PM10:  9.10
PM2.5: 1.80
------
End reading SDS011
Call sensorDHT
Start reading DHT11/22
DHT22 couldn't be read
DHT22 couldn't be read
DHT22 couldn't be read
DHT22 couldn't be read
DHT22 couldn't be read
------
End reading DHT11/22
Call sensorBME280
Start reading BME280
Temperature : 0.00 C
humidity : 0.00 %
Pressure    : 0.00 hPa
------
End reading BME280

Vielen Dank für die Unterstützung!

Adorfer, 2017-07-24:
BME280 funktioniert hier einwandfrei, die Messwerte passen auch zum ebenfalls angeschlossenen DHT22. 

Bitte prüfe mal, ob der Sensor nicht versehentlich an 5V (UBE oder wie immer es auf dem Board heisst) hängt. Wenn dem so sein sollte, dann könnte er davon "ableben" (oder bereits bei einem Fehlersuche gestorben sein, zumindest habe ich das mir einem hinbekommen) .

PaulchenPlump, 2017-07-24:
Hallo Adorfer,

danke für deinen Hinweis, aber ich bin mir ziemlich sicher, dass der Sensor korrekt angeschlossen ist. Hier noch ein Bild auf dem Sensor und Board mit den entsprechenden Kabeln verbunden sind: [https://picload.org/view/rwgddccr/img_20170724_070725.jpg.html](https://picload.org/view/rwgddccr/img_20170724_070725.jpg.html). Unglücklicherweise sind die Kabelfarben nicht ideal und die anderen Sensoren stecken noch dran, aber man kann die Verkabelung erkennen...

Im Thread [105](https://github.com/opendata-stuttgart/sensors-software/issues/105) gibt es ja ähnliche Probleme, aber da wollte ich mich nicht direkt mit einhängen, da dort noch anderes im Argen liegt...

dokape, 2017-07-24:
Ich habe die 3.3v für den BME aus der linken Reihe, der 5.  oder so PIN von unten genommen. 
Ansonsten sieht das aber anhand des Bildes gut aus.
Mehr als die 4 Kabel brauchts ja auch nicht. 
Hast du evtl. beim Einlöten der Stiftleiste eine Brücke oder schlechte Lötstelle eingebaut?
Evtl ein Kabel defekt?

ricki-z, 2017-07-24:
Unter https://playground.arduino.cc/Main/I2cScanner gibt es ein Programm, mit dem man den I2C Bus scannen lassen kann. Dort sollte der BME280 dann auf jeden Fall auftauchen, wenn er funktioniert. Eventuell haben deine Exemplare eine Bus-ID, die wir nicht abfragen.
Zu ändern ist in dem Programm lediglich Wire.begin() in Wire.begin(D3, D4), dann kann die Verkabelung bleiben.

PaulchenPlump, 2017-07-24:
Den 5. Pin von unten habe ich auch schon probiert - leider ohne Erfolg. Soeben habe ich noch einmal die Kabel und den Sensor getauscht ebenfalls ohne Erfolg. Eine Lötbrücke will ich ausschließen, kalte Lötstellen bei zwei Sensoren eigentlich auch. :)

Ich habe Testweise auch den I2CScanner installiert und dort findet er ebenfalls nichts. :( Habe deshalb versucht den Sensor an andere D-Pins anzuschließen, aber er findet einfach nichts... Dann hatte ich noch einen zweiten NodeMCU den ich für zukünftige Bastelarbeiten aufgehoben habe. Aber: ihr ahnt es schon - wieder ohne Erfolg. :(

Was kann ich noch probieren? Gibt es Software-Seitig irgendetwas zu aktivieren (abgesehen von dem einen Schalter für den BME280 auf der config-Seite)? I2C oder irgendwelche Zusatzbibliotheken für den Upload?

Leider habe ich bisher weder mit dem NodeMCU noch mit Arduino etwas gemacht, aber der Start war echt simpel - nur jetzt komme ich an meine Grenzen.

ricki-z, 2017-07-24:
Zumindest mit dem I2C Scanner sollte sich die BME280 bestimmen lassen. Wenn das nicht geht, geht irgendwo etwas grundsätzlich schief. Der Scanner macht nichts anderes, als einfach alle I2C-Geräte-IDs anzusprechen und auf eine Antwort zu warten. Da muss nichts konfiguriert oder aktiviert werden. Und ohne die Bibliotheken würde sich der Scanner gar nicht übersetzen lassen.

Adorfer, 2017-07-25:
Hast Du evtl einen RasPi an dem Du das mal testen kannst? 
Ich tippe fast auf eine Tüte defekte Sensoren. 
Oder irgendwas mit "spiegelverkehrt angeschlossen", mehr ist wirklich nicht übrig

dokape, 2017-07-25:
Am Sensor ist der BME nur mit den 4 Kabeln anzuschließen, dann USB-Kabel zur Stromversorgung dran, auf der config-Seite aktivieren, und Sensor neu starten. 
Dass es bei 2 BMEs und 2 nodeMCU schief läüft... Sehr komisch.

PaulchenPlump, 2017-07-25:
OK, mittlerweile gehe ich auch von einer "Tüte defekter Sensoren" aus. Sind definitiv Pin SCL und SDA bei diesem Sensor zu verbinden (CSB und SDO bleiben leer)?! 
Wo habt ihr eure Sensoren her - würde es dann nochmal mit zwei neuen probieren.  Oder kann ich auch ohne RasPi die Funktion testen (habe nur noch irgendwo einen Odroid3)?

Vielen Dank!

Adorfer, 2017-07-25:
Kannst Du mal ein Bild Deiner Verdrahtung posten? (z. B. via imgur).
 Vielleicht übersehen wir einen Aspekt.

Adorfer, 2017-07-25:
Von den RPI-Leuten hörte ich, dass die BME280  durchaus nach wenigen Tagen (im geschlossenen Gehäuse, im Betrieb, Eigentümer keiner Schuld bewusst) sterben können  Mir wurden auch defekte angeboten ("habe hier ne Hand voll"). Von daher: Die Qualität scheint nicht konstant zu sein.

PaulchenPlump, 2017-08-13:
Ich wollte euch die Rückmeldung geben, dass ein neuer BME280 die Lösung brachte. Löten und Einbau 10min und vorher habe ich mich Stunden gequält. :( Aber nun ist alles gut!

Vielen Dank für eure Unterstützung und als Warnung an andere: Ja, es werden offensichtlich auch defekte Sensoren verkauft!

closed 2017-08-13 by PaulchenPlump

ottowayne, 2018-03-04:
Habe nun von zwei verschiedenen Verkäufern von AliExpress BME280 Sensoren gekauft und beide funktionieren nicht. Kann es wirklich sein, dass es so viel Ausschuss bei den Sensoren gibt?

ricki-z, 2018-03-05:
Kannst du ein Bild der Sensoren posten? Wäre interessant, ob es da vielleicht eindeutige Merkmale gibt. Ich bestelle größere Mengen der BME280 und habe Ausfälle im unteren einstelligen Prozentbereich. Die kosten aber auch ca. 3,50 Euro bei AliExpress. Bei Angeboten unter 2 Euro würde ich vom Kauf abraten.

Adorfer, 2018-03-05:
solche mit "onboard 3,3V-Regler" (die also mit 5V versorgt werden dürfen) scheinen tendenziell besser zu sein.