opendata-stuttgart/sensors-software #111: /values should show time/date of values

jms72, 2017-08-05:
For the current values under /values it should be shown from what time these values were collected (if there's an issue with absolute time, it's perfectly fine to say "xx seconds/minutes ago")

Adorfer, 2017-08-10:
this implies the sensor knows a valid time. 
(i do not know if keeping an ntp client is really neccesary on a device which does not store any data locally anyway)

ricki-z, 2017-08-10:
The NodeMCU doesn't have a RTC and NTP is crashing with the rest of our firmware. So we can only print something like "xx seconds/minutes ago". I will try to insert this in one of the next versions.

ricki-z, 2017-08-13:
The next version will show the time since last measurement under /values and as an entry in /data.json .

ricki-z, 2017-08-17:
New Version is online. Sensors will auto update, if this isn't disabled. Please look at the values page and the /data.json if this is okay for you.
Don't forget to close the issue f it's okay.

jms72, 2017-08-18:
Thanks so much for the quick turnaround. Due to travels I will only have
opportunity to check in about three weeks.

jms72, 2017-09-03:
Confirming it works great

closed 2017-09-03 by jms72