opendata-stuttgart/sensors-software #112: Improve in case of WiFi down - cache some of the values

jms72, 2017-08-05:
The sensor should try to reconnect in regular intervals if WiFi is not available when starting or lost. (it looks like the WiFi SSID created by the sensor is supposed to only be around for a limited time anyway.)

In case of WiFi not being enabled, sensor values should be cached as much as memory allows.

ricki-z, 2017-08-07:
The software tries to reconnect if the connection is lost. The wifi AP mode is only active for 10 minutes.
We can't save any values because of the limited RAM of the NodeMCU. There are some function calls where we don't know the RAM usage (http transmit). So we could get a crash if we save too many values. And me need the available space for future sensors.

dokape, 2017-10-09:
@jms72 
Are your questions answered? 
Wifi-behavior is as designed. In config you can change the duration of Accesspoint. Standard is 600sec, -> 10 Minutes. Should be enough to setup the login for your own wifi-network. 

Hint: Android devices connected to the airrohr-network disconnects because they can not route to internet and reconnects to your home-Wifi with internet :-( 
So sometimes it's easier to delete your standard Wifi-connection on Android devices to avoid a disconnect. 

Can this issue be closed?

jms72, 2017-10-11:
Thanks.
I might go ahead and implement caching values on device - in a cloned
repository then. Even one value per sensor per hour would be valuable and
would definitely fit in memory for say 12 hours of WiFi not being
available. I could also envision an algorithm reducing precision over time
with length of no signal.

no-username42, 2017-10-17:
There seem to be versions with 4 and up to 16mb flash https://en.wikipedia.org/wiki/ESP8266 available.

Storing data locally would be very useful for 'mobile' sensors.
a) sensors that are located in a polution hotspot with no available wifi nearby.
b) sensors that are on the move to measure the polution that pedestrians, bikers and joggers are actually exposed to - inbetween cars for example, where no stationary sensor can be located.

Adorfer, 2017-10-17:
having an GNS  ("GPS board") would provide exact time. And a micro-SD could perfectly store data for weeks. 
Perhaps we start running out of IO-Pins now... SD-cards need another 4 wires.

jordanc365, 2019-07-12:
Has anyone had any progress saving values to an SD card? 
Also how long will the sensor attampt to recconect to wifi before it gives up and needs a reboot?

DeeKey, 2019-12-30:
@dirkmueller please add a FeatureRequest tag.
Yey with working NTP ESP8266 can get exact time and store the data in user space to transmit data in the future.
Also I think that solution of this issue also requires changes on the server side.