opendata-stuttgart/sensors-software #116: Pressure scale at madavi.de/sensor

pauledd, 2017-08-09:
Hi

I am just curious about the x-scaling of the pressure values for my BME280.
You can see it at:
https://www.madavi.de/sensor/images/sensor-esp8266-11732458-bme280-3-day.png
It seems to stick quite close to 1k. Is this a scaling issue or might my sensor defect? I ask because
the pressure made some bigger jumps the last days as I saw on my local wheater station.

Adorfer, 2017-08-09:
So just another defect BME280. 
Of course not, it must be a sw issue, or what do you imply? 
So what is tue current value if you read it out? is there even any?
A graph with **without scale** on on the x looks bogous. Or is it 0 mbar at y-axis?
In this case either the sensor is in some kind of preassure chamber. or it's simply defect.
(https://github.com/opendata-stuttgart/sensors-software/issues/109 looked like defect BME280 )

pauledd, 2017-08-10:
I currently removed the sensor to test it separately, so dont wonder about the up-spike. And of cause I meant y-axis in my opening post, sorry for that. I was simply strange to see no other values than "1k" on the y-axis. So it was not clear to me what min-max range I see there. Could be 900-1100hPa or also 999,99-1000,01 hPa.

ricki-z, 2017-08-10:
The y-axis starts at zero. This should be changed in the next weeks to see the smaller changes.

pauledd, 2017-08-10:
thank you and my sensor is okay.

closed 2017-08-10 by pauledd