opendata-stuttgart/sensors-software #121: Config-Änderungen ersetzten Auth-Passwort

lemmingz, 2017-08-19:
Werden nachträglich Änderungen an der Config gemacht (bspw. das Hinzufügen der OpenSenseMap-ID), wird durch das Speichern das Basic-Auth-Passwort auf x-mal * gesetzt, wobei x für die Anzahl der Zeichen des ursprünglichen Passworts entspricht.

Der Fehler liegt hier in der form_password-Funktion die das Passwort durch Sternchen ersetzt. Da dieser Wert dann aber der Value des Inputs ist wird das Passwort entsprechend geändert.

ricki-z, 2017-08-20:
Wenn das Passwort nur aus Sternchen besteht, wird es nicht neu gesetzt. Dadurch sollte das von dir beschriebene Verhalten eigentlich nicht vorkommen. Siehe dazu die Zeilen 1128 und folgende.
Damit soll verhindert werden, das man aus dem Konfig-Formular das Passwort im Klartext auslesen kann.

lemmingz, 2017-08-20:
Ja, für das WLAN-Passwort findet der Check statt, aber nicht für das www_password. Ich habe mich damit gestern für eine Weile vom Sensor ausgesperrt.

ricki-z, 2017-08-20:
Alles klar, danke für den Hinweis. Wird gerade berichtigt. Alle anderen Passwörter sind nur Inputfelder. Dort war die Maskierung und Überprüfung nicht nötig.

ricki-z, 2017-08-20:
Die neue, korrigierte Version NRZ-2017-096 geht gerade raus. Die meisten Sensoren sollten diese in 24 Stunden haben. Bitte testen und wenn der Fehler damit entfernt ist, den Issue bitte schließen.

lemmingz, 2017-08-21:
Version NRZ-2017-096 kam an und der Fehler tritt nicht mehr auf. 

Vielen Dank!

closed 2017-08-21 by lemmingz

ricki-z, 2017-09-06:
@lemmingz Kannst du die Version NRZ-2017-098 testen (sollte in den kommenden Stunden auf deinem Sensor landen)? Bei mir hat es auf 2 Sensoren funktioniert, wie es sollte. Aber vielleicht habe ich irgendeine Möglichkeit übersehen.

lemmingz, 2017-09-19:
Sieht soweit gut aus. (NRZ-2017-099)