opendata-stuttgart/sensors-software #124: discontinuous JSON query

pauledd, 2017-08-24:
Hi
I periodically query JSON data from my device via the command "curl -s http://192.168.1.150/data.json".
I do this every 5 minutes. Throughout the day I get multiple episodes where I simply get empty responses from the device. 
I can eliminate WLAN/Network/Sensor error because the sensor continues sending values to your servers while the gaps happen, gaps are not visible on your "madavi.de/sensor" graphs.
Here is some example csv data that I write into a logfile. The last three values are from the BME180 sensor and you can see there is a big gap. That happens several times a day.

```
2017-08-24 08:25:14,22.4,1001.1,18.2,61.8
2017-08-24 08:30:18,22.5,1001.1,18.4,60.9
2017-08-24 08:35:22,22.5,1001.2,18.4,60.5
2017-08-24 08:40:27,22.6,1001.1,18.6,59.9
2017-08-24 08:45:37,22.7,,,
2017-08-24 08:50:47,22.8,,,
2017-08-24 08:55:58,23.0,,,
2017-08-24 09:01:08,23.2,,,
2017-08-24 09:06:19,23.3,,,
2017-08-24 09:11:29,23.5,,,
2017-08-24 09:16:39,23.6,,,
2017-08-24 09:21:50,23.6,,,
2017-08-24 09:27:00,23.7,,,
2017-08-24 09:32:11,23.7,,,
2017-08-24 09:37:15,23.8,1000.9,21.9,50.4
2017-08-24 09:42:19,23.8,1000.7,22.0,49.8
2017-08-24 09:47:23,23.8,1000.8,22.6,49.2
```

Any ideas anyone?

ID: 11732458
Firmware: NRZ-2017-097

pauledd, 2017-08-25:
It seems there is some kind of pattern in the data retrieval failure. There is a big gap every day from around 21/22 o'clock to 7 o'clock in the morning. 

Complete log of the last three days:
https://pastebin.com/raw/E9a3vUjp

ricki-z, 2017-08-25:
But the SDS011 values are shown?

pauledd, 2017-08-25:
when I query json and it fails I get nothing back, but at the same time the webinterface shows all values correct.

pauledd, 2017-08-25:
I will try to write a retry routine in my query bash script that tries again after 30 secs if the query was empty. Maybe that helps.

ricki-z, 2017-08-25:
We had to disable the internal webserver while sending data to the APIs. The NodeMCU has too little RAM to handle both at the same time. But this should only affect some seconds per interval (2.5 minutes).

pauledd, 2017-08-25:
Well, I query every 5 min the json data, maybe that sometimes jumps into the 2.5 min api interval, I'll watch how it behaves with my 30s retry, currently I had no retry. If that doesnt help I'll change my 5 min to something else.

pauledd, 2017-08-26:
I found the problem, its not the firmware, it was my stupidity. For testing porposes I connected my RPi that runs the scripts via a network bridge to my desktop client and of cause it got no data when I switched my desktop of. I re-enabled the wifi stick on the RPi and everything is good. Sorry for the issue...

closed 2017-08-26 by pauledd