opendata-stuttgart/sensors-software #127: "An eigene API senden" funktioniert nicht

simonszu, 2017-08-29:
Hi,

ich hatte Interesse daran, die Feinstaubdaten mittels node-red in MQTT umzuwandeln, und habe mir diesbezüglich mal die Funktion "an eigene API senden" angesehen. Da ich hierzu keinerlei API-Dokumentation gefunden habe, habe ich einfach mal mitgesniffed, was der Sensor hier für Daten raushaut. Ergebnis: nichts. 

Daher eigentlich 3 Issues auf einmal:

- Es fehlt eine Dokumentation, was für eine Art API der Sensor erwartet
- Der Sensor macht trotz aktiviertem Häkchen keinerlei Requests auf dem definierten API-Endpoint
- MQTT-Integration wäre ein super Feature, was dem Sensor definitiv noch fehlt. 

Ich weiß, dass ich die Daten wieder aus der luftdaten-Datenbank rausholen kann, aber dies möchte ich aus zweierlei Gründen nicht: Zum einen erscheint es mir unsinnig, die Daten erst in der Cloud zu parken, um sie dann wieder herauszuholen, wenn zwischen dem Sensor und meinem Computer nur 4 Meter Luftlinie sind, und zum anderen: Meine Daten gehören mir. Ich finde euer Projekt super, und bin gerne teil der Datenerhebungs-Crowd, aber die Daten, die der Sensor misst, nur so umständlich wieder zu bekommen hinterlässt ein ganz kleines schales Beigeschmäckle. Ich finde die InfluxDB-Integration schon super, aber so richtig MQTT wäre halt noch ein Stück geiler.

dokape, 2017-08-29:
Auf meinem lokalen raspi funktionierts mit diesem script:

https://github.com/opendata-stuttgart/madavi-api/blob/master/data_simple.php

Schreibt alle Daten vom Sensor in tägliche .csv Dateien. 
Hatte es kurz über flogen, das sollte auch mit kleinen Änderungen in deine lokale Datenbank schreiben können.

geektoor, 2017-08-29:
Ahoi,

ich hab selbst mal damit rumgespielt und erst den selben Eindruck gehabt, wie simsonzu schildert. Der erste und vermutlich wichtigste Hint war, das die Daten via. php://input rein kommen, und nicht etwa als Variablen oder so.
Allerdings, wenn simsonzu sich nicht ungluecklich ausgedrueckt hat, scheint es ja gar keinen Request zu geben. Das waere in der Tat komisch. Evtl. noch mal genau die Details checken, vlt. ist irgendwo ein Typo. Ausserdem auf die Konsole gucken, der Sensor loggt sinnvoll mit (9600 Baud).

Bzgl. MQTT gibt es schon mehrere Issues dazu und die Aussage das es aus verschiedenen (nachvollziehbaren) Gruenden in naher Zukunft kein MQTT gibt. Ein relativ einfacher Grund ist der begrenzte Speicher des NodeMCU (braucht ja auch Platz fuers OTA Update), ein anderer ist die verschiedenen Implementierungen von MQTT, mal mit Client IDs, mal ohne, dafuer mit Authentifizierung usw.
Daher ist die eigene API eigentlich als "neutrales Gateway" eine echt gute Wahl. Man kann sie ja als Proxy bauen und die Daten damit dann z.B. an ein MQTT weitergeben - vermutlich war das auch Deine Idee, simsonzu? :)

Gruesse,
Sven

ricki-z, 2017-08-29:
Hallo simonszu,

zu Punkt 1: Wir organisieren das Projekt in unserer Freizeit (könnte das bitte irgendwann mal zur Konntnis genommen werden?). Deshalb ist die Doku nicht an allen Stellen optimal. Aber der Sourcecode unserer Software ist frei zugänglich und sollte mit mittleren Programmierkenntnissen zu lesen sein. Der Aufwand, die gesuchte Information zu finden, sollte sich in Grenzen halten.
zu Punkt 2: Der Sensor macht trotz aktiviertem Häkchen keinerlei Requests auf dem definierten API-Endpoint
Der Sensor macht einen Request auf den Endpunkt. Ein Blick ins Access- bzw. Errorlog des angesprochenen Webservers würde das zeigen. Dein 'Sniffen' kann kein Sniffen gewesen sein, sonst hättest du den POST-Request gesehen. Die Daten werden als JSON-'Datei' gepostet und sind deshalb z.B. nicht in den normalen $_POST-Variablen in PHP zu sehen.
zu Punkt 3: In den Issues sind auch zwei 'Work in Progress' Issues zu finden. In Issue #33 habe ich begründet, warum wir MQTT nicht unterstützen werden.

Zum Punkt 'Zugriff auf die Sensordaten direkt vom Sensor': Der Sensor stellt die Daten unter http://<ip-des-sensors>/data.json zur Verfügung. Damit müssen die Daten ebenfalls nicht erst das eigene Netz verlassen.

simonszu, 2017-09-03:
Jo. Tatsächlich habe ich den Sniffer irgendwie falsch bedient. Dass es ein POST-Request sein soll hat mir schon mal viel weitergeholfen. Ich habe jetzt einen Node-Red-Workflow fertiggestellt, der als "Eigene Api" für den Sensor dienen kann, und die ganzen Daten auf einem MQTT Broker ablädt. Einen entsprechenden Blogpost werde ich schreiben, sollte dann also irgendwann auf Google zu finden sein :)

closed 2017-09-03 by simonszu

cpietsch, 2020-01-30:
Hey @simonszu besten Dank für deinen [Workflow](https://blog.simonszu.de/luftdaten-mqtt-gateway-fuer-den-feinstaubsensor/) - habe den gleichen Usecase mit mqtt.
Bei mir kommt nix im node-red an. Was genau hast du im Sensor eingestellt ?
Bei mir sieht das so aus: 
[image]
()

simonszu, 2020-01-30:
Bei mir siehts genau so aus. Hast du eventuell noch irgendwo http Auth für den http endpoint node gesetzt? Was passiert, wenn du einen curl request gegen den node machst?

cpietsch, 2020-01-30:
Hi, der endpoint im node-red geht - hatte ich per curl getestet. Schätze der Sensor sendet die Daten nicht richtig. Gibt es ne Möglichkeit an die Logs vom Sensor zu kommen ohne Serial ?

simonszu, 2020-01-30:
Du kannst die serielle Konsole von nem esp auch irgendwie über TCP ansprechen, aber frag mich nicht wie.

ricki-z, 2020-01-30:
Aktuell geht das Auslesen der Debug Infos nur über Serial.
Nimm mal das http:// beim Server raus, dort sollte nur der Name oder die IP stehen. Ob http oder https genutzt wird, wird über die Checkbox rechts über dem Feld bestimmt.

cpietsch, 2020-01-31:
@ricki-z das klingt so, als wenn der Fehler daher kommt. Hatte ne 2017 Firmware drauf und jetzt hat er sich auf die neuste geupdated und das alte Default so mitgenommen.

cpietsch, 2020-01-31:
Kann bestätigen, dass das "http://" den Fehler verursacht hat. Die Daten kommen jetzt - juhu!
```
{"esp8266id":"1497234","software_version":"NRZ-2020-129","sensordatavalues":[{"value_type":"SDS_P1","value":"3.00"},{"value_type":"SDS_P2","value":"2.70"},{"value_type":"samples","value":"906753"},{"value_type":"min_micro","value":"32"},{"value_type":"max_micro","value":"20273"},{"value_type":"signal","value":"-87"}]}
```