opendata-stuttgart/sensors-software #133: Zähler: "XX Sekundem seit der letzten Messung" zählt falsch rum

dokape, 2017-09-08:
Version NRZ-2017-098

Zähler: "XX Sekunden seit der letzten Messung" zählt falsch rum

Der Sekundenzähler zählt von 145 bis auf Null. (Vorherige Firmwares: von 0 bis 145)
Ergibt mit dem Text "Seit der letzten Messung" keinen Sinn.
Zählrichtung wieder aufwärts oder Text ändern: "XX Sekunden bis zur nächsten Messung"

Die Sekunden werden auf ganze Sekunden gerundet und dann formatiert mit zwei Nachkommastellen .00 ausgegeben. Da die Nachkommastellen immer .00 sind, konnte man die auch wieder entfernen.

dokape, 2017-09-08:
Interessant: der Wert "age" der "data.json" zählt korrekt aufwärts! 
Also ein reines Darstellungs-Problem?

ricki-z, 2017-09-08:
Ja, war ein Copy-n-Paste Fehler. Ich wollte die zu hohen Werte wegbekommen, die während oder kurz um das Senden angezeigt wurden. Leider habe ich die Änderungen kopiert und 'vergessen', das die beiden ja in unterschiedlicher Richtung zählen. Update kommt.

dokape, 2017-09-11:
NRZ-2017-099
Sieht gut aus. Zählt wieder in die richtige Richtung.

*closed*

closed 2017-09-11 by dokape