opendata-stuttgart/sensors-software #134: Feature request: add servo driving functionality

intui, 2017-09-08:
I'd like to attach a servo Motor to drive an arrow visualizing the Feinstaub value and attach it to the house so it is visible for the cars that drive by.

dokape, 2017-09-08:
What about a stand-alone solution? 
An additional nodeMCU with servo-driver that pulls the values by json-data from the sensor? 
Try yourself: 
http://<Your-Sensor-ip>/data.json

ricki-z, 2017-09-08:
We have to focus on functionality needed more than once. The available flash space is very limited. By now we use 86 percent of (firmware) flash and 59 of RAM.

dokape, 2017-09-11:
is there any requirement for more discussion or can this be closed?

intui, 2017-09-21:
ok. to close. I'll try the standalone approach. Thanks

closed 2017-09-21 by ricki-z