opendata-stuttgart/sensors-software #137: BME 280 install description

sbamueller, 2017-09-17:
Provide a description on how to connect the BME 280 to the ESP

dokape, 2017-09-17:
There is a german description at the FAQ:

dokape, 2017-09-19:
Die Readme.md wurde entsprechend erweitert. 
https://github.com/opendata-stuttgart/sensors-software/blob/master/airrohr-firmware/Readme.md

Besteht noch Diskussionsbedarf dazu?
Wenn nicht, könnte man das issue schließen.

Adorfer, 2017-09-21:
2 points: 

- BME280 vs BMP280: BMP lacks the hygrometric sensor, ebay&co offers are often written for both types, but off course the BMP280 is shipped
- BME280 is far supperior than DHT22(AM2302) in terms of hygrometric tolerance, in "best case", and espcially there are series of DHT22 which go completly crazy when "above 50%".  See https://www.kandrsmith.org/RJS/Misc/Hygrometers/calib_many.html

So: BME280 is my clear recommendation, eventhough it does not look nearly as nice.

dokape, 2017-09-22:
I have a DHT22 and a BME280 in the same sensor. The temp-Values differ 2°C, the DHT is lower
The hygrometer differs also, The BME is higher. Both in same Case. 
See Screenshot.
I'm still not shure which sensor has the better values...

[image]

dokape, 2017-09-22:
BTW: Can this issue be closed? 
Due to topic "Install description" the documentation with relevant information is provided.

Adorfer, 2017-09-22:
for low values, DHT22 and BME280 seem to correlate, for higher it's most likely that DHT22 is completly nuts. 
My recommendation would be, to drop DHT22 as a default in the instruction, and replace it by BME280. 

(sensors here glued "back to back" on the same cable

[image]
[image]
[image]

[image]

[image]

pathmapper, 2017-09-22:
@Adorfer thanks for sharing!

Even if BME280 and DHT22 would display the same values, I would favor BME280 because of the additional value air-pressure. The only drawback is that you have to solder the BME280, the price is only insignificant higher. 

It is nice to have support of all the different kind of sensors, but it is also a little bit confusing.
Therefore it would be helpful to have a recommendation which sensor works the best and this should be the default.

A little bit off topic in this issue, but it would be good to have a place to gather all information related to the accuracy and reliability of the all the supported different sensors.
See also https://github.com/opendata-stuttgart/feinstaub-map/issues/49

dokape, 2017-09-22:
DHT22 can every user connect to the nodeMCU. 
BME280 has to be soldered: the posts for a cable are not soldered to the board.

Adorfer, 2017-09-24:
As "BlueberryE BME280" you get it with a a standard 4.2mm connector, where normal jumper cables fit. 
(but off course that's not 4€, but around 10€)

https://www.amazon.de/BlueberryE-Breakout-Board-BME280/dp/B01EV2E2V0/
http://blueberrye.cc/ppt/blueberrye-breakoutboard-bme280/

[image]

pathmapper, 2017-09-24:
@Adorfer 

At the bootom of this page http://blueberrye.cc/ppt/blueberrye-breakoutboard-bme280/ is written:

> Please Note contrary to several other BME280 breakout boards this one uses I2C adress 0x76 instead of 0x77. Take care if you use libraries.

Is the "BlueberryE BME280" compatible with the airrohr-firmware?

Adorfer, 2017-09-25:
In der Tat ist die Adresse derzeit in der Adafruit-Library hardcoded: 

https://github.com/adafruit/Adafruit_BME280_Library/blob/b2fd691dc77fed0a6f0487dfc0a837bda88eb99f/Adafruit_BME280.h
Zeile 32
    #define BME280_ADDRESS (0x77)

Also was tun? 
Der Sensor BME280 ist ja durchaus sinnvoll hinsichtlich "Luftdruck" und "um Meilen zuverlässigerer Hygrometer-Werte. 

Ich kann anbieten, an 50 (oder meinestwegen 200) Sensoren (die für unter 4€ vom China-Versender) die Pfostenleisten einzulöten. Und zum Selbstkostenpreis löte ich auch Kabel mit Pfostensteckern gewünschter Länge dran und tunke alles (bis auf den Sensor) zweimal in Epoxy, also ein paar Cent. 
Nur auf den Versand an Einzelpersonen habe ich wenig Lust. Falls also irgendein Space das übernehmen würde.

janzieg, 2017-10-13:
Concerning BME280 values (which are different then those from DHT22):
When using the Addafruit-Driver with its default configuration, the temperature values are 1-2 degrees too high! (and dependent other values are "wrong")

You can also check: http://www.esp8266.com/viewtopic.php?f=13&t=8030&sid=5b736d896eb9758f6ae56ce5aefbaceb&start=28

So BME280 has some self heating depending its sampling configuration mode.

This can be optimizied by setting the "oversampling" during BME280 init (refer to spec https://ae-bst.resource.bosch.com/media/_tech/media/datasheets/BST-BME280_DS001-11.pdf - "weather monitoring").

The Adafruit driver can enable this by calling "setSampling()". This should be integrated into Airrohr-firmware - afterwards the values will not be that different to DHT22.

Adorfer, 2017-10-13:
@janzieg 

> This should be integrated into Airrohr-firmware - afterwards the values will not be that different to DHT22.

Sounds like an relevant issue. Do you like to open a new one? (i could do, but this might look like "stealing")

janzieg, 2017-10-13:
@Adorfer https://github.com/opendata-stuttgart/sensors-software/issues/147

Adorfer, 2017-10-23:
@pathmapper 

  >  Is the "BlueberryE BME280" compatible with the airrohr-firmware?

Now it is

[image]

pathmapper, 2017-10-24:
@Adorfer cool, thanks for sharing!
According to your screenshot it is already implemented in the current firmware release NRZ-2017-099, right?

Adorfer, 2017-10-24:
@pathmapper correct. 
I was just not aware about that, but that is the current debug output on the serial console.

dirkmueller, 2019-11-27:
The bme280 sensor is only operated in force measurements mode in firmwares from 2018 or newer, so the overheating should be gone. Closing as fixed.

closed 2019-11-27 by dirkmueller

kwaaak, 2020-05-18:
Default pin-out (Lieferzustand)
[image]