opendata-stuttgart/sensors-software #139: compile error with undeclared GPIOS

poesel, 2017-09-18:
for software version NRZ-2017-100 - code unchanged

this is a much too obvious error so I must be doing something wrong but I can't figure out what

Thanks!

error log:
`Arduino: 1.8.3 (Mac OS X), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck, Serial, None"

WARNUNG: Bibliothek LiquidCrystal_I2C behauptet auf (avr) Architektur(en) ausgeführt werden zu können und ist möglicherweise inkompatibel mit Ihrem derzeitigen Board, welches auf (esp8266) Architektur(en) ausgeführt wird.
airrohr-firmware:204: error: 'D3' was not declared in this scope
 SSD1306   display(0x3c, D3, D4);
                         ^
airrohr-firmware:204: error: 'D4' was not declared in this scope
 SSD1306   display(0x3c, D3, D4);
                             ^
In file included from /Users/markus/Documents/Arduino/airrohr-firmware/airrohr-firmware.ino:113:0:
ext_def.h:82: error: 'D1' was not declared in this scope
 #define SDS_PIN_RX D1
                    ^
/Users/markus/Documents/Arduino/airrohr-firmware/airrohr-firmware.ino:213:26: note: in expansion of macro 'SDS_PIN_RX'
 SoftwareSerial serialSDS(SDS_PIN_RX, SDS_PIN_TX, false, 128);
                          ^
ext_def.h:83: error: 'D2' was not declared in this scope
 #define SDS_PIN_TX D2
                    ^
/Users/markus/Documents/Arduino/airrohr-firmware/airrohr-firmware.ino:213:38: note: in expansion of macro 'SDS_PIN_TX'
 SoftwareSerial serialSDS(SDS_PIN_RX, SDS_PIN_TX, false, 128);
                                      ^
ext_def.h:135: error: 'D5' was not declared in this scope
 #define GPS_PIN_RX D5
                    ^
/Users/markus/Documents/Arduino/airrohr-firmware/airrohr-firmware.ino:214:26: note: in expansion of macro 'GPS_PIN_RX'
 SoftwareSerial serialGPS(GPS_PIN_RX, GPS_PIN_TX, false, 128);
                          ^
ext_def.h:136: error: 'D6' was not declared in this scope
 #define GPS_PIN_TX D6
                    ^
/Users/markus/Documents/Arduino/airrohr-firmware/airrohr-firmware.ino:214:38: note: in expansion of macro 'GPS_PIN_TX'
 SoftwareSerial serialGPS(GPS_PIN_RX, GPS_PIN_TX, false, 128);
                                      ^
ext_def.h:52: error: 'D7' was not declared in this scope
 #define DHT_PIN D7
                 ^
/Users/markus/Documents/Arduino/airrohr-firmware/airrohr-firmware.ino:222:9: note: in expansion of macro 'DHT_PIN'
 DHT dht(DHT_PIN, DHT_TYPE);
         ^
ext_def.h:127: error: 'D7' was not declared in this scope
 #define DS18B20_PIN D7
                     ^
/Users/markus/Documents/Arduino/airrohr-firmware/airrohr-firmware.ino:247:17: note: in expansion of macro 'DS18B20_PIN'
 OneWire oneWire(DS18B20_PIN);
                 ^
/Users/markus/Documents/Arduino/airrohr-firmware/airrohr-firmware.ino: In function 'String sensorPPD()':
ext_def.h:70: error: 'D6' was not declared in this scope
 #define PPD_PIN_PM1 D6
                     ^
/Users/markus/Documents/Arduino/airrohr-firmware/airrohr-firmware.ino:2379:23: note: in expansion of macro 'PPD_PIN_PM1'
   valP1 = digitalRead(PPD_PIN_PM1);
                       ^
ext_def.h:71: error: 'D5' was not declared in this scope
 #define PPD_PIN_PM2 D5
                     ^
/Users/markus/Documents/Arduino/airrohr-firmware/airrohr-firmware.ino:2380:23: note: in expansion of macro 'PPD_PIN_PM2'
   valP2 = digitalRead(PPD_PIN_PM2);
                       ^
/Users/markus/Documents/Arduino/airrohr-firmware/airrohr-firmware.ino: In function 'void setup()':
airrohr-firmware:2734: error: 'D3' was not declared in this scope
  Wire.begin(D3, D4);
             ^
airrohr-firmware:2734: error: 'D4' was not declared in this scope
  Wire.begin(D3, D4);
                 ^
In file included from /Users/markus/Documents/Arduino/airrohr-firmware/airrohr-firmware.ino:113:0:
ext_def.h:70: error: 'D6' was not declared in this scope
 #define PPD_PIN_PM1 D6
                     ^
/Users/markus/Documents/Arduino/airrohr-firmware/airrohr-firmware.ino:2760:10: note: in expansion of macro 'PPD_PIN_PM1'
  pinMode(PPD_PIN_PM1, INPUT_PULLUP); // Listen at the designated PIN
          ^
ext_def.h:71: error: 'D5' was not declared in this scope
 #define PPD_PIN_PM2 D5
                     ^
/Users/markus/Documents/Arduino/airrohr-firmware/airrohr-firmware.ino:2761:10: note: in expansion of macro 'PPD_PIN_PM2'
  pinMode(PPD_PIN_PM2, INPUT_PULLUP); // Listen at the designated PIN
          ^
exit status 1
'D3' was not declared in this scope

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.
`

ricki-z, 2017-09-18:
The 'error' for LiquidCrystal_I2C is only a warning. The lib is also working for NodeMCUs.
You need to change the board to 'NodeMCU 1.0 (ESP-12E Module)'. This will include the definitions for the GPIOs.

poesel, 2017-09-18:
*facepalm* obviously! 
Thanks! :)

closed 2017-09-18 by poesel