opendata-stuttgart/sensors-software #143: Visualize data on display

nobels, 2017-09-24:
I got this idea to visualize data on this small display.
This is the list of data that we can show

- Tempareture
- Humidity
- SDS PM2.5
- SDS PM 10
- GPS Lat & Longitude -> (optional)
- IP address -> if you are not connected with USB cable, you can not detect IP address
- Wifi signal strength -> bad or good connection
- Wifi Network -> (optional)
- Sensor firmware -> because you can auto update it, you can keep track of it.
- Measurement Counter -> This is the total count off measurments send. If device reboots you will notice because the value is not increasing.

Now i should set the display timer interval to 3 or 5 seconds to switch screens like mockups below.

The text can be (re)used from the translationfiles :-)

What i also like is that you can choose what displays you would like to see.

[image]

dokape, 2017-09-25:
actually there is work to improve display data on the display.
keep in mind, there is also the classic LCD 1602 with I2C port supported.

nobels, 2017-09-25:
The focus here was for the Display Module 0.96 inch IIC Serial Blue OLED

I know there is a different screen -> but I don't know the limitations of the screen.
I gues it will be 2 lines and how many characters are there in 1 line?

I was thinking to add position text like 1/6, 2/6 -> 6/6

ricki-z, 2017-09-25:
Hi dokape, hi nobels,
I would like to get a discussion about the information that has to be shown on the displays first. So the shown screens are a good start. The implementation on different displays is the second step.
The LCD1602 has 16 cols on 2 rows. So the first and second display can show nearly the same as on the OLED. The other screens could be split in a sligthly different way.

nobels, 2017-09-25:
Ok, i already made a list. 
Can someone add something that i forgot ?

dokape, 2017-09-25:
Nobels idea look good. 
Every some seconds a new screen rotating through. 
similar the screens on the 1602 could go through. 

I would provide a "start screen" showing the software-version. 
would it make easy to discover if flashing a new version manually was successful
or if the old version still is on ;-)

The actual firmware support following hardware:

-  SDS011 (Feinstaub)
-  PMS1003, PMS5003, PMS6003, PMS7003
-  PMS3003
-  DHT22 (Temperatur, rel. Luftfeuchte)
-  HTU21D (Temperatur, rel. Luftfeuchte)
-  PPD42NS
-  BMP180
-  BMP280
-  BME280
 - DS18B20
 - GPS (NEO 6M)

We do not know, which hardware is connected on a specific "airrohr". 
When we display the SDS-PMs, there could be a PPD or a PMS-Sensor and what should we display.
Or me. e.g. I have a DHT22 AND a BME280 installed. Which data should odr would I see? 
So only the activated sensors should be shown. But you have to create a screen for every sensor (template)!
Would it be senseless to integrate a second checkbox for "Show Data on Display?"

Make the titles international. So the language-files should be recognized. 

My similar issue #101 to this is here: 
https://github.com/opendata-stuttgart/sensors-software/issues/101
I will close it an discuss it here.  

I wait for a 1602 and a 2004 LCD-Screen for delivery.
Should we define a milesstone for the integration?

ricki-z, 2017-09-25:
For different hardware with same values: We will define a priority, i.e. temperature of a BME will override temperature of DHT. Same for other sensors: PPD42NS -> PMS -> SDS.
But normally the nodes will have only one sensor per type. Comparing values should be made via archived data, '/data.json' or the integrated webserver.
Anything else will blow up the firmware.

dokape, 2017-09-25:
when going that way: would it be possible to show on one screen:

Starting up: 
Firmware version (1st screen)
trying Connection to WIFI xxx (2nd screen)
Not established: No Network. Connect me at "SSID" and "192.168.4.1"
established: 
Show IP-Adress during lookup for new Software. 
download ...   flash...     |   or: No new firmware.

Wating until first messure (like on webserver, counting down to zero)

Show values (do we have a date/time?

Date/Time
PM 2.5     13µg    
PM 10      15µg
Temp:       13,4 °C
rel. Luftf    87,4%
?(Luftdruck  1027hPA)

Wobei wir beim Luftdruck wieder bei der Diskussion #102 mit Abweichungen beim Aufstellort / Höhe wären.

Adorfer, 2017-09-25:
I would like to have a "slave ESP", one in can mount inside with the LCD. 
A nodeMCU that gets (via what method whatsoever) the data from her outdoor sister.
And a would not mind a "show-mode" to connect to multplie outdoor sensors... (downstairs street, roof) to cycle through...
But of course i have no idea how they should find each other, if the ip adresses are not static and the slave display does not want to query it "form the internet".

dokape, 2017-09-26:
A slave with Display for inside I would like to have.
Well, that could be an own project independent Form the Firmware. 
A static IP should be necessary in first steps. 
My opinion: Keep this in mind for a next milestone and discuss the visualization of data here.

ricki-z, 2017-10-03:
A slave with a display is a completely different project. For this you could use other devices than an esp8266. A simple html page with some javascript should work (using the json from the sensor or our API).

nobels, 2017-10-03:
@dokape  There is a nice app for that called dustify for android. That is a solution for your question. (but this is off topic)

dokape, 2017-10-06:
[image]

Ein LCD 2004 bietet deutlich mehr Platz als das 1602. 
Wäre das evtl auch zu unterstützen?

Adorfer, 2017-10-06:
@nobels  " There is a nice app for that called "Dustify" for android. That is a solution for your question."
maybe there is an android app, but on my ESP8266 i can't install them. 
(And i have no intention to glue an additional "old phone" with a permanently glowing screen on the kitchen wall. a second 'airrohr' with display, but without the sensor, would be perfect for me, but if that's not possible, then it's a separate project)

ricki-z, 2017-10-06:
LCD 2004 kommt, warte noch auf mein Testexemplar. Dann sollte aber fast Schluß sein mit Displays.
Die aktuelle Beta hat eine geänderte Anzeige mit mehreren 'Bildschirmen'. Falls jemand testen will ...

dokape, 2017-10-07:
Testversuch:
Branch Master: keine Änderung seit 15 Tagen  (kompiliert zu Version NRZ-2017-100, keine mehrere Bildschirme)
Branch Beta: keine Änderung seit 19 Tagen.

Download https://www.madavi.de/sensor/update/data/latest_beta.bin
ergibt eine Version NRZ-2017-099 die jedoch keine mehrere "Bildschirme" hat.

Hab ich falsche Quellen?

ricki-z, 2017-10-07:
Sorry, hatte den Sync vergessen. Jetzt sollte Beta aktuell sein.

dokape, 2017-10-07:
Ah.. da.
Kompilierung läuft durch. 
Tipp: Bei Version zur Unterscheidung evtl. beta1, beta2 etc anhängen.  Hab ich mir für mich eingefügt,
um zu sehen, ob flashen auch funktioniert hat.

Beide Displays laufen fehlerfrei durch. 
Sieht echt gut aus. insbesondere am O-LED. Fast genau so wie @nobels vorgeschlagen hat. 

Die Frage ist: reicht es nicht, die Wifi-Info und Device-Info, nur 
zu Beginn anzuzeigen? evtl. bis zur ersten Messung? und dann nur noch die Messwerte?
Cool ist, dass die Anzahl der Messungen angezeigt wird.

Der DHT22 wird derzeit noch nicht angezeigt?

dokape, 2017-10-07:
Fehler in der Beta: 
Im Web werden die Messdaten jetzt mit 2 Nachkommastellen angezeigt.

ricki-z, 2017-10-07:
Habe es gerade auf 1 Nachkommastelle geändert. Version endet jetzt mit '-b2', spätere dann aufsteigend. Es kommt auch noch ein 'Forced firmware update', dann könnten die Versionsnr. ohne Endung bleiben und die 'Beta-Tester' können entscheiden, ob sie 'updaten' wollen.
Das ist alles noch Vorbereitung auf die Auto Beta Updates ...

ricki-z, 2017-10-07:
Die Temperatur-, Luftfeuchte- und Luftdruck-Sensoren werden nacheinander durchgegangen und der jeweils 'Beste' angezeigt. Bei DHT22 und BMP180 würde also die Temperatur vom BMP180 kommen, die Luftfeuchte vom DHT22 und der Luftdruck vom BMP180.
WiFi-Info und Device-Info sind auch zwischendurch ganz praktisch. Der Sensor läuft ja unter Umständen schon eine Weile, wenn man mal die IP-Adresse braucht. Dann muss man den nicht neu starten.

nobels, 2017-10-07:
Is it possible to talk in English please in general? (This avoids copy paste time in google translate) ;-)
And also because you would like to cover Europe with Luftdaten :-D

ricki-z, 2017-10-07:
I've pushed a new version to the beta branch. This version includes some of your ideas (multiple screens on oled, counting measurements,...) Changes to release version are listed in der versions.md.

dokape, 2017-10-07:
Fast changes. Cool.
Knowing the priority of shown values will be interesting for the docu.
Writing in english with german spell check on the tablet is arrgl... So I'm sorry for switching to german.
Will test tomorrow.

nobels, 2017-10-07:
https://stackoverflow.com/questions/20197421/adding-a-second-dictionary-to-the-spell-check-in-android-4-3

dokape, 2017-10-08:
NRZ-2017-100-B2 compiles

**Web:**
1 decimal OK
New: waiting until first messure: values shown as 0, old behavior: nothing.

**Displays:** 
only DHT22 activated: Headline of screen shows: "DHT22 /" Why a slash?
(The BME 280 is connected, but not activated)

**Differences** between oled and LCD:**

_Screen: Device Info:_
LCD shows:   IP and WIFI Network  (wrong)
O-LED shows:  ID and Firmware (correct)

_Screen: WIFI Info:_
LCD shows:  ID and Firmware  (wrong)
O-LED shows:  IP and WIFI-Network  (correct)

ricki-z, 2017-10-08:
@dokape 
NRZ-2017-100-B3:
changes made according to your list
added units on oled display
some minor changes to save flash space

dokape, 2017-10-09:
Thanks. really fast your changes! ;-)

Looks good. 

2 Remarks:
LCD: could it be possible to add µg/m³ to the PM-Values as on O-LED? 
Not shure if this characters are supported by display.

O-LED and LCD:
if one PM-Value has  is like 5.5 and the other is like  12.5, it's left-justified like:
2.5
12.5
Could it be possible to right-justify the values? (Even Check for other values Temp, Humidity)

_Titles:_
As the complete FW is language located, the titles for Values and headlines should also be
in the language files. 
Device Info -> Geräte info
Hum: -> Feuchte?
Pres: -> Druck?
Measurements -> Messungen
etc.
Even the possiblilty of different languages should be integrated. I'm not shure if the translation 
should be done at this time. As the complete firmware supports it, it should be done even for the displays ;-)

Tests to be done:
Start with vanilla configuration... 
Tests with other hardware (Checked with SDS011, BME280, LCD 2004 instead of 1602, O-LED, DHT22),
I'm waiting for the HTU and NEO-Hardware. 
Test with DS18B20

dokape, 2017-10-09:
Question:

The construct with 

 if ((next_display_count % 4) == 0) {

could only display 4 screens.  ==0, ==1, ==2, ==3. 
To add screens in future, the complete construct has to be changed? Well, every if-clause
has to be changed  from  if ((next_display_count % 4) == 0) { to  if ((next_display_count % 5) == 0) { or more.

example:
```
Case next_display_count
case 1: screen 1
case 2: screen 2
case 3: screen 3
case 4: next_display_count=0 (reset)
next_display_count ++
```

This should be easier to add new screens? 
Just my thoughts....

ricki-z, 2017-10-12:
Language location will be done if the display function is more or less feature complete. So all translations can be done at once.
We may have different numbers of screens for the oled and the lcd displays. 
The informations shown on the displays should be configurable. I.e. if the display is on a public place the network information may have to be disabled.  So the whole 'if's for screen selection may change. I have to look for the 'best' solution ... 
(sorry, very chaotic answer)

dokape, 2017-10-13:
BUG: HTU 21D values not shown on Display

Software NRZ-2017-100-B3
connected Sensors:
DHT22
BME280
HTU21D

All activated: Web: OK. all values shown.

deactivating DHT22 and BME280, only HTU21D activated:
No display of Temp and Hum. on O-LED and LCD
HTU21D-values are shown on Web.

dokape, 2017-10-18:
Build Beta 4 Test:

HTU21D is shown. 
DS18B20 not testet.
right align: looks good. OK.     description/readme: Left-align (should be right align)

nobels, 2017-10-22:
My remarks with beta 4
- I have HTU21D and there is a presure line -> no values (No need to show, because it does not have this feature)

And what i should do is add more spaces between the paging  not this  . . o .    but this-->      .    .    o    .

I think there are max 4 or 5 pages.

But i like the result ! ;-)

dokape, 2017-10-30:
GPS-Data are not shown on Displays.

dokape, 2017-11-06:
NRZ-2017-100-B6

OLED:  More spaces provided.

dokape, 2017-11-11:
NRZ-2017-100-B7:

GPS-Data (NEO 6M) are shown on displays.
LCD: Due to 2 lines: only lon and lat are shown.
O-LED: lon, lat, alt is shown.

Issue on OLED: 

the . . o . . 
ist shown twice. See pic.

[image]

dokape, 2017-11-13:
NRZ-2017-100-B8:
screen GPS: OK

It seems, this implementation runs fine now. 

I would suggest not to implement more at the moment and provide this features 
in the next master-version.

dokape, 2017-11-14:
NRZ-2017-100-B9:

Wifi-quality (Signal) now same value as in web/value-page. Thanks.

When Wifi not reachable, the information of Sensor-AP are shown on OLED.

stewecar, 2018-01-26:
Hello,
Sorry if i am missing something..how it is possibile to use an OLED display simultaneously with a bme280? They use the same pins..
Thanks for any advice!
stefano

stewecar, 2018-01-26:
Hi, thanks a lot for the reply; so, i have to use a beadboard cause there is no way, i think, to use the same pins of the ESP..
Thanks again!
Stefano

ricki-z, 2018-06-09:
@nobels @dokape The beta changes are now part of the actual release version. Can we close this issue (and open a new one if any new wishes will pop up)?

saschaludwig, 2018-06-15:
Hi, nice work! In what branch or tag do I find the above multi-page changes?
It would be nice if there was a timeout for the oled display. Oleds don't last forever.

Also, a button for waking up the display/advancing pages would be nice. I added a touch Sensor to my mobile GPS based Sensor here:

[image]

Obicom, 2019-03-06:
Solved it with a new compilation and an additional command ... but maybe as a feature request for a new version .. would be great if it would be implemented in the WebUI as an option.

[image]