opendata-stuttgart/sensors-software #144: plötzlich keine Daten mehr

dokape, 2017-09-25:
Firmware NRZ-2017-099

produktiver Sensor war einige Tage problemlos durchgelaufen.
Doch plötzlich lieferte er keine Daten mehr.
Die Webseite war nicht erreichbar. 
Pingen konnte ich den Sensor jedoch. 
Nach Stromlos machen lief wieder alles.

Letzte von mir vorgenomme Maßnahme:
Ich bin auf die Konfigurationseite gewesen und hatte im Browser die Sensoren markiert und hier in ein Issue kopiert. Da lebte die Software noch. Kurz drauf war sie dann weg. 
Komisch.

dokape, 2017-09-26:
Heute morgen wieder ein "komischer" Effekt:
Beim Aufruf der /value Seite erscheint die Mitteilung, dass noch xx sekunden bis zur ersten Messung sind.
5 Minuten später bei der Kontrolle schon wieder.
Dann lief der Sensor ca 1-2 Stunden,
Jetzt ist er wieder nicht erreichbar. Pingen ist möglich. 
Der eigene Webserver ist nicht erreichbar. 
Daten werden auch keine abgeliefert, der lokale Raspberry pi webserver schreibt keine Daten mehr in die lokale .csv-datei. 
Auf Maps.luftdaten.org auch keine Daten. 
Opensensemap ebenfalls keine Daten.

Bug in der Firmware oder ist meine nodeMCU hinüber? 
Oder ein Problem im Netzwerk trotz pingbar?

dokape, 2017-09-26:
.csv-dateien durchforstet.
Seit dem 19.9.2017 haben meine lokalen .csv-dateien nur noch größen unter 50kb. 
Import nach Excel und Kontrolle bringen eine Anzahl von Datensätzen deutlich unter den üblichen 580 Einträgen pro Tag. 
Unregelmäßige Aussetzer zwischen 3 Minuten und 30 Minuten sind vorhanden. 
Oft nur ein Aussetzer pro Stunde. 

Die Firmware NRZ-2017-099 ist jedoch seit 2017-09-10 online.

dokape, 2017-09-27:
Nun ist mein Airrohr gestern um 22:00 ganz hängen geblieben. 
- Keine Daten abgeliefert
- per http nicht erreichbar.
- Pingen geht (hatte ich über Nacht im 10s takt durchlaufen, nicht ein Ping ins Leere.

Nach einem Stromlos-Reset war er sofort wieder pingbar und per http erreichbar. 
Daten werden abgeliefert. 

Hardware-Fehler (nodeMCU) oder Software-Fehler?

dokape, 2017-09-27:
Andere nodeMCU vorbereitet und getestet.

Dort auch nach 2 Messungen Absturz bzw. Hänger.
Ausgabe am seriellen Port beim debuggen:

`# Sending to opensensemap: 
Start connecting to ingress.opensensemap.org
Requesting URL: /boxes/59c144d6-SENSEID-b5001/data?luftdaten=1
526279
{"software_version": "NRZ-2017-099", "sensordatavalues":[{"value_type":"SDS_P1","value":"10.27"},{"value_type":"SDS_P2","value":"7.83"},{"value_type":"temperature","value":"22.10"},{"value_type":"humidity","value":"77.50"},{"value_type":"BME280_temperature","value":"23.65"},{"value_type":"BME280_humidity","value":"53.15"},{"value_type":"BME280_pressure","value":"97513.12"},{"value_type":"samples","value":"579044"},{"value_type":"min_micro","value":"231"},{"value_type":"max_micro","value":"286156"},{"value_type":"signal","value":"-53"}]}`

danach war Stille.

Der Fehler trat bei mir ab 19.9. auf. 
Seit 19.9. liefere ich auch Daten an eine opensensebox

Ich habe nun die API OPENSENSEBOX deaktiviert und lasse die bisherige nodeMCU dran.

Bin gespannt obs an der Datenlieferung nach Opensensebox liegt.

dokape, 2017-09-27:
Die API für Opensensemap ist nun seit 8h deaktiviert.
Der Sensor hat jede 2 Minuten brav ohne einen einzigen Ausfall seine Daten abgeliefert.
Die max:_cycle sind sind seit der Deaktivierung wieder alle im Bereich 27 bis 28000 msec. 

=> Weiter Suchen nach Fehler in API mit OpenSenseMap

dokape, 2017-10-01:
3 Tage ohne aktivierte API OpenSenseMap und kein einziger Fehler bzw. Ausfall.

sandboxgangster, 2017-10-06:
Moin, 
Meine hat auch hin und wieder mal ausfälle, hab es bei mir mit einem neuen Netzteil lösen können, das China ding war wohl nicht so toll, vielleicht hilft das bei dir.

Ansonsten zicken dir Dinger auch gerne bei Äußeren Einflüssen Gewitter z.b. das sieht man auch gut auf der Karte das bei Gewitter die Sensoren in der Umgebung verschwinden ist auch kein wunder gibt keine Abschirmung, PLC stört auch, meins hängt sich auch auf wenn ich die Mikrowelle benutze ... ;)

dokape, 2017-10-06:
Zusammenhang klar mit Opensensemap.

Heute endlich mal Zeit gehabt, etwas näher zu schauen.

Testsystem mit NRZ-2017-100 (beta, aktueller Entwicklungsstatus)

Sobald ich die API für Opensensemap aktiviere, bleibt die Software genau beim Übertragen der
Daten hängen. Anbei ein DEBUG INFO vom seriellen Monitor:

Die Daten kommen bei Opensensemap noch an und werden dort angezeigt. 
Das LCD-Display zeigt in diesem Moment auch nur noch Murks an.
Der Sensor ist immer noch pingbar! 

----------------------
Checksum is: 251 - should: 251
PM10 (sec.) : 4.70
PM2.5 (sec.): 2.00
End reading SDS011
Start reading SDS011
Checksum is: 251 - should: 251
PM10 (sec.) : 4.70
PM2.5 (sec.): 2.00
End reading SDS011
Start reading SDS011
Checksum is: 251 - should: 251
PM10 (sec.) : 4.70
PM2.5 (sec.): 2.00
End reading SDS011
Start reading SDS011
Checksum is: 251 - should: 251
PM10 (sec.) : 4.70
PM2.5 (sec.): 2.00
End reading SDS011
Start reading SDS011
Checksum is: 252 - should: 252
PM10 (sec.) : 4.80
PM2.5 (sec.): 2.00
End reading SDS011
Start reading SDS011
Checksum is: 253 - should: 253
PM10 (sec.) : 4.90
PM2.5 (sec.): 2.00
PM10:  4.72
PM2.5: 2.00
------
End reading SDS011
Start reading DHT11/22
Humidity    : 48.80%
Temperature : 23.00⸮C
------
End reading DHT11/22
Start reading BME280
Temperature : 23.57 C
humidity : 44.18 %
Pressure    : 973.06 hPa
------
End reading BME280
Creating data string:
WLAN signal strength: -75 dBm
------
output values to display...
## Sending to opensensemap: 
Start connecting to ingress.opensensemap.org
Requesting URL: /boxes/59c144d6d67eb5001107aace/data?luftdaten=1
11729146
{"software_version": "NRZ-2017-100", "sensordatavalues":[{"value_type":"SDS_P1","value":"4.72"},{"value_type":"SDS_P2","value":"2.00"},{"value_type":"temperature","value":"23.00"},{"value_type":"humidity","value":"48.80"},{"value_type":"BME280_temperature","value":"23.57"},{"value_type":"BME280_humidity","value":"44.18"},{"value_type":"BME280_pressure","value":"97305.80"},{"value_type":"samples","value":"608084"},{"value_type":"min_micro","value":"217"},{"value_type":"max_micro","value":"406380"},{"value_type":"signal","value":"-75"}]}
--------------  letzte Ausgabe -------------------------

dokape, 2017-10-16:
ARGL. War nun ein paar Tage problemlos durchgelaufen.
Nun habe ich komplette nicht-Erreichbarkeit des Sensors. 
Immer mehr Ausfälle seit 3 Tagen, Ping nicht möglich. 
Irgenwann kommt der Sensor wieder, liefert 3-10 Mal Daten und ist dann wieder eine halbe Stunde nicht erreichbar. 
Softwarefehler? Hardware-Fehler? Netzwerkproblem? Ich kanns nicht eingrenzen. :-/

markbeee, 2017-10-16:
In meinem Fall hatte ich Probleme mit dem Feinstaubsensor, der immer wieder mal keine Daten lieferte. Da die Verbindungen alle nur mit eher minderwertigen Kabelbrücken (jumper wire) ausgeführt sind, könnten die bei hoher Feuchtigkeit/ Kälte etc. eine mögliche Fehlerquelle sein. Ich habe darauf hin alle Verbindungen mit Drähten fest verlötet. Seitdem (~ein paar Tage) habe ich keine Probleme mehr.

dokape, 2017-10-16:
Das wäre durchaus möglich.

dokape, 2017-10-18:
Oder doch Netzwerkprobleme? Heute morgen konnte ich mich mit keinen Device auf den Repeater, an dem auch der Sensor hängt, Verbindungen. Irgendwo bei der Zuteilung der (festen) Ip-Adressen hängen geblieben. Einstellungen dazu im Router geändert, nun läuft es wieder besser. Am WE mal die Kabel rauswerfen und das Zeug auf einer Platine sockeln und stecken.

dokape, 2017-10-25:
Bastelstunde am Wochenende:
Alles auf eine Trägerplatine gebaut.
nodeMCU gesockelt und BME280, DHT22 per Sockelleiste gesteckt. 
SDS011 per orignalem Flachbandkabel angesteckt. 

Aber das größte Problem war ganz fies.
Ein durchgeführtes Routerupdate. Neue Firmware verhindert die Vergabe von IP-Adressen von Devices, die über den Repeater angeschlossen sind. 
IP-Adress-Anforderung kommt am Router an, der bietet dem Device eine Ip-Adresse an.
Ein DHCPACK jedoch wird erst nach 100ten Versuchen durchgeführt. Läuft dann ne halbe Stunde, dann
gehts wieder von vorne los. 

Was aber am ursprünglichen Problem mit der Lieferung von Daten an OpenSenseMap und dem Hänger nichts zu tun hat.

Adorfer, 2017-10-25:
Sorry, but this is totally different topic
> Ein DHCPACK jedoch wird erst nach 100ten Versuchen durchgeführt.

Welcome to the KRACK aftermath. 
See comments like: https://github.com/freifunk-gluon/gluon/issues/1240#issuecomment-338744575

(And consider putting your sensor in a a local Freifunk-Mesh)

dokape, 2017-10-25:
Here is no Freifunk-Mesh.
All firmwares (airrohr, Repeater and the routers firmware) are dated before KRACK known.

Adorfer, 2017-10-27:
This not about freifunk or mesh, but about "reliability of UDB based protocolls in networks with high packet loss due to whatever reason. Like the use of repeaters, which even have issues in WDS mode due to hidden-station effets .
DHCP is one of the first victims. And the next are clients with less robust protocoll stacks are even more prone to suffer if they are hopping constantly between the (home gateway's) AP and WDS/MIFI AP(s). 
Or if their carrier sensing is not fully functional.

dokape, 2017-10-29:
My last problem is a problem between actual router firmware and actual repeater-firmware. older versions match together.  but thats not the problem here.

still a problem to transfer data to opensensemap. it causes the airrohr to stop and reboot. 
several nodeMCU tested, also without repeater...

kristigu, 2018-04-06:
How did you solve the problem? I have the same issue more or less.. I can access the data locally and read the measurements but I see that after some time the data doesn't get uploaded online anymore. I also think the problem started more or less after I configured Opensensemap.

MkMunich, 2018-05-06:
hi, I also have the problem, that data is not sent to "api-rrd.madavi.de". After 24h I normally get a mail and then I have to restart the ESP8266 (by repowering). Did someone solved the problem?

dokape, 2019-04-09:
FW -123b
nicht mehr nachvollziehbar.

Issue closed.

closed 2019-04-09 by dokape